hathach/tinyusb #502: Warm plug fail

Thanos-Love-Ironman, 2020-09-01:
- **PC OS**   : Windows 10/ 
 - **Board**   : stm32h743 ,my own board
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**
My project could not support USB warm unplugged since my board doesn't use VBUS sensing pin.(alternative function)
It is only working at the first plug when my board is not powered by USB cable.
Any solution???

duempel, 2020-09-04:
Do I understand correctly that your board uses its own power supply and is not bus powered? Maybe this problem is related to #179 . I've also had problems with disconnecting and reconnecting without reseting the controller but had no time to search for the cause yet. Are you using FS or HS peripheral?

Thanos-Love-Ironman, 2020-09-05:
> Do I understand correctly that your board uses its own power supply and is not bus powered? Maybe this problem is related to #179 . I've also had problems with disconnecting and reconnecting without [...]

Yes,I use FS peripheral.My DTR bit cannot be set by pc terminal.I don't know if it's relative to this issue.
But now I got an another issue.My cdc port will crash after transfer data for couple minutes if the usb task priority is low.
It return in  here : 
     TU_VERIFY( !usbd_edpt_busy(TUD_OPT_RHPORT, p_cdc->ep_in), 0 );
which means the busy flag is miss to set false or there is once the data transfer fail and didn't triger the DCD_EVENT_XFER_COMPLETE.I am using FreeRTOS.
I am about to rewrite the busy setting logic,I hope it will work.

hathach, 2020-09-05:
I don't know any other way to detect disconnection even with STM32 except for the current https://github.com/hathach/tinyusb/blob/master/src/portable/st/synopsys/dcd_synopsys.c#L1001 

Most of stm32 devkit support this though. Maybe you should bring this issue to ST community, their engineer could possibly know the answer.

duempel, 2020-09-10:
@Thanos-Love-Ironman I see you had some other problems which seem to stick together. Right now the information I have don't seem that there is actually a failure to usb because of warm plugging.

Does your Windows detects the USB device correctly (even after 2nd or 3rd plug)? Then the enumeration seem to work fine without any problems. The problem would be related to other system setups I guess.

But if the enumeration process will fail it could be because of the alternative function on USB sensing pin. I don't know your exactly controller but other STM32 that uses OTG seem to have a hardware comparator fixed to one pin. This comparator will connect the integrated FS pull-up to the dataline. If for example a output function will pull this pin low, I could imagine that the pull-up is not set correctly and Windows will notify you about problems to the usb device. Maybe this pin is changed during your runtime and that's causing the different behaviour between first and n-st plug.