hathach/tinyusb #503: usb_device_task Priority set up

Thanos-Love-Ironman, 2020-09-03:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   :  Windows 10
 - **Board**   : stm32h743
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**
I am using latest ST sd card HAL Driver .I found out I have to set up usb_device_task with a high priority.Otherwise the TinyUSB won't work if I read sd card and open cdc port at the same time.

Thanos-Love-Ironman, 2020-09-05:
I find out that if the USB IRQ is running fast than the usb task,the cdc or msc will crash very soon.But I cannot set a high priority with usb task,because there are another tasks which cannot be delay.So，have TinyUSB take this into consideration？Is there any workaround?
Thank you!

hathach, 2020-09-05:
I don't see any information that I could analyze and help you with. If you want people feedback, you should fist capture the rtos task switching with freertos trace, segger system view etc ... and post the screenshot here. Also provide the your own analysis & its context such as where the is the time the issue happen and why the CPU is blocked on certain resource.

Thanos-Love-Ironman, 2020-09-06:
> I don't see any information that I could analyze and help you with. If you want people feedback, you should fist capture the rtos task switching with freertos trace, segger system view etc ... and [...]

I really don't know what's the detail,I only found out it will stuck in OTG_FS_IRQHandler() infinite loop.

Thanos-Love-Ironman, 2020-09-07:
> I don't see any information that I could analyze and help you with. If you want people feedback, you should fist capture the rtos task switching with freertos trace, segger system view etc ... and [...]

the reasons is usbd_edpt_busy(TUD_OPT_RHPORT, p_cdc->ep_in) always return false after open terminal couples of minutes.

hathach, 2020-09-07:
you could take a look at this #507 for how to analyze and troubleshoot your issue. Since you are the only one that could reproduce the issue. These are mostly race condition, and these required to capture the whole picture of resource conflict between running task to be able to determine the root cause.

Thanos-Love-Ironman, 2020-09-08:
> you could take a look at this #507 for how to analyze and troubleshoot your issue. Since you are the only one that could reproduce the issue. These are mostly race condition, and these required to [...]

I don't think this is the cause,I have rewrite the  FIFO with xStreamBuffer.There is mutex when call tud_cdc_n_write_flush in other high prioity task.
The MSC is also crash just like the cdc,I have to set the usb_task priority very high.
[image]

Thanos-Love-Ironman, 2020-09-08:
> you could take a look at this #507 for how to analyze and troubleshoot your issue. Since you are the only one that could reproduce the issue. These are mostly race condition, and these required to [...]

The MSC will crash soon even when I use no cdc function( no mutex).If I use cdc funtion at the same time,it will crash sooner.

Thanos-Love-Ironman, 2020-09-08:
> you could take a look at this #507 for how to analyze and troubleshoot your issue. Since you are the only one that could reproduce the issue. These are mostly race condition, and these required to [...]

it will stuck in here as well (debug) when I read sd card and open pc terminal.
[image]

hathach, 2020-09-08:
github issue is not forum support, it your custom code. You should be able to troubleshoot it.

closed 2020-09-08 by hathach

Thanos-Love-Ironman, 2020-09-08:
> github issue is not forum support, it is your custom code, please try to troubleshoot it yourself.

this issue  exists when I use unchaged TinyUSB.It's a bug

Thanos-Love-Ironman, 2020-09-08:
> github issue is not forum support, it is your custom code, please try to troubleshoot it yourself.

I have try to fix this for weeks,I have try to diasble the usb interrupt after every transfer and enable it in usb task.this will solve the cdc crash issue but msc crash issue.

hathach, 2020-09-09:
It is not about the modification of tinyusb, it is your whole project. What you mentioned is not reproducible by anyone. And you literally didn't provide any information for troubleshooting/anallyzing it. **You could try to file another issue for this problem with**
- A simplest modification to stock example to reproduce the issue. It is best to reproducible on one of the supported bsp. This makes sure other could reproduce it. Else It could be your own bug in other part of the project.
- Provide all the template information, including step to reproduce, debug log etc..
- More in-depth analysis where/why the issue occurs with screenshot when needed. Many including me normally won't try to run a custom example unless I feel it is quick enough to fix or it is severe issue that need to fix asap.

As I said, try to file issue similar to what I did in #507 . Or like this one https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/565, the user surely have much more complicated sketch, but they could simplify it so that It can be reproduced easily. And please don't just bump up the issue like a forum, debugging is the lengthy process, we don't have to know everything of it. Put a sum up when you really stuck since I am not interested in debugging another project. I have too many works already.