hathach/tinyusb #504: Convert EP buffers to ring buffers (at least for ISO EPs)

PanRe, 2020-09-03:
Hello,

Right now @kasjer and i are working on finalizing a working version for an audio 2.0 driver. During tests a possible bottleneck came up and i wanted your opinion on this issue. In general a FIFO is necessary in order to decouple the sampling process from the USB transmission process. 

So the setup right now (e.g. for a microphone) is: Samples -> FIFO -> EP buffer -> USB FIFO, where -> means "put into".

As how tinyUSB is set up right now, all EP buffers are considered to be linear buffers, however, at least for ISO EPs, ring buffers (or FIFOs if you like) would be far more practical. The benefits would be:

- Less memory space required as the EP ring buffer would not need to be as large as the current FIFO plus the linear EP buffer
- One unnecessary copy process less
- The ring buffer can be filled by use of a DMA in circular mode with little intervention (picture an I2S interface which directly moves its samples to USB, where the only thing to do would be to update the write pointer of the ring buffer adequately)

However, this would mean a rather big amount of work to do since all implementations of dcd_edpt_xfer() must be adapted. A possible compromise would be to setup a new function exclusivly for ISO EPs e.g. dcd_edpt_ISO_xfer() which gets a pointer to a FIFO. Within the device controler drivers, we could check if the EP is an ISO EP and if so read the data from there.

All reasons stated above also hold in case of OUT EPs.

What do you think of this?

Talking about ring buffers:

For the current FIFO implementation (which are ring buffers) i would also suggest to change from relative to absolute pointers (or unmasked indices), which is very nicely described [here](https://www.snellman.net/blog/archive/2016-12-13-ring-buffers/). The benefits are:

- You don't loose one byte since empty and full are distinguishable from the unmasked or absolute pointers
- An overrun can be determined (also nice for DMA use)

In the article, the editor says that ring buffers with unmasked pointers can only work for buffer lengths which are a power of two (and wrap around of unsigned integers must be possible). However, i think i came up with a solution to this problem (making any length possible). So if you are interested please let me know!

Also in your current implementation of the FIFOs i would recommend you not to store a count variable and modify it in both pull() and push() as the count variable might get messed up if two processes modify it (the writing of the count variable is not atomic). Rather i would compute count all time new.

Thanks for your opinions!
Reinhard

hathach, 2020-09-05:
> Hello,
>
> Right now @kasjer and i are working on finalizing a working version for an audio 2.0 driver. During tests a possible bottleneck came up and i wanted your opinion on this issue. In general a [...]

Thank you (both of you) very much for working on the ISO and suggestion on the DMA + ring buffer. I think it is better for ISO to have its own dcd xfer API() if needed. ISO is always an special xfer in most device/host controller anyway, its API can actually have more arguments than normal one.  

Regarding DMA and endpoint buffer, there is a gotcha since not all MCUs could support DMA, some have dedicated USBDATA regsiter to read/write to, another could do DMA but requiring the input buffer address must be aligned with 64 bytes (NXP LPC series) and/or within a specific SRAM address which is DMA-abled for USB controller. The EPBuf  is mainly for this purpose (CFG_TUSB_MEM_SECTION/CFG_TUSB_MEM_ALIGN). However, don't be bother by those limitation, you could just submit a draft PR with stub ( no need for actual implementation just yet). I could help to spot those cross-mcu issue.

> Talking about ring buffers:
>
> For the current FIFO implementation (which are ring buffers) i would also suggest to change from relative to absolute pointers (or unmasked indices), which is very [...]

Thanks for the article, it is a great read. It does ring a bell to me, I think I did look at this approach before, but deciding that power of 2 limitation is probably too much. Since I am not entirely how the stack would use the FIFO, and may want to have a as small as possible fifo to save space on small mcus.  But it doesn't seem to be a problem so far. 
 
> Also in your current implementation of the FIFOs i would recommend you not to store a count variable and modify it in both pull() and push() as the count variable might get messed up if two [...]

yeah, the count is unnecessary since it is can be inferred by write/read/depth. I was too lazy to do the math previously. Regarding the atomic, it is rather safe I guess, since it is wrap within fifo mutex lock/unlock when using with preempted RTOS. For noOS, fifo write is only called for the device queue task, which is guarded by  interrupt enable/disable. But I could mis some cases with this. 

Overall, I am open to all the suggestion and approach, as long as it doesn't break user/application API (which isn't in this case) and should be able to run all on the MCUs which can be troublesome due to mentioned limitation.

cr1901, 2020-09-05:
Even if DMA is supported on a microcontroller, my opinion is that tinyusb shouldn't hard-require implementations (at _least_ the non-iso part of the user API) to use DMA. Tinyusb should remain relatively easy to port, and I think hard-requiring DMA complicates this.

I'm not even sure if hard-requiring DMA is wise for iso xfers (STM32F4 and friends come to mind- they support iso, but I recall that nothing says you _have_ to use DMA. And STM should be fast enough to process interrupts).

PanRe, 2020-09-05:
Oh i never ment to hard require a DMA for ISO EPs. Most often, a DMA can not be used at all e.g. in my case I have 4 microphones and I need to encode them into a PCM stream with the four samples interleaved. This (at least for STM32F4) can not be done by DMA. Converting the EP buffer into a ring buffer, however, would give you a nice opportunity to attach a DMA (in case of an I2S stream where stereo samples are already interleaved).Of course the user API shall not be broken!--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

hathach, 2020-09-07:
> Also in your current implementation of the FIFOs i would recommend you not to store a count variable and modify it in both pull() and push() as the count variable might get messed up if two [...]

I just remembered why I use count instead of computing it, since I don't want to waste 1 slot of storage, which could be 8/16 bytes depending on the item_size, the count variable is always be 2 though. And it is less math to do. Read index, write index and count are modified next to each other within mutex guarded code. I hope that would be enough for race condition.

PanRe, 2020-09-08:
> I just remembered why I use count instead of computing it, since I don't want to waste 1 slot of storage, which could be 8/16 bytes depending on the item_size, the count variable is always be 2 [...]

Yeah i see what you mean but unfortunately, this implementation as it is right now is not ISR save as you finally use xSemaphoreTake() within your tu_fifo_lock() (which according to the docs must not be called from an ISR). Thus the cnt state is not save in this condition. However, if the plan was to use this FIFO as an EP buffer and the content of the buffer gets copied into the USB buffer from within an ISR (as it is right now in dcd_synopsys.c) then we have a problem.

As i mentioned before, if you use unmasked write and read pointers you do not suffer from the lost memory slot at all (which is really nice). Then, you also would not need the cnt state any more to evade the loosing slot problem. This would also make the FIFO invurnable to ISR calls since there is no state any more which gets modified from both read() and write() functions. The write function only modifies (write to) the write pointer and read() only the read pointer. Reading the pointers to determine the count state is never a problem. The only thing i am uncertain right now is what happens with the mutexes in case i want to read from the FIFO from within an ISR...
And yes, the price for this would be to calcuate the count state from new every time it is needed. But this is neglectable i guess. Oh and you further save the two bytes saving the count state at all ^^.

hathach, 2020-09-08:
@PanRe yeah, I found the unmasked approach is nice as well. Though the power of 2 limitation can be a bit too strict for now. Maybe we can go back to this later on when everything is figured out. Since I don't see much benefit to implement it for now. 

Within TinyUSB, the mutex API won't ever be called within ISR, the only osal that invoked in ISR is `osal_queue_send(osal_queue_t qhdl, void const * data, bool in_isr)` with the sole purpose to put the usb event into the usbd task queue for processing. User also shouldn't call any TinyUSB API() in a ISR as well (maybe read-only API could be safe there). It has been designed as such from the start. 

> And yes, the price for this would be to calcuate the count state from new every time it is needed. But this is neglectable i guess. Oh and you further save the two bytes saving the count state at all ^^.

These computation is nothing :), the power-of-2 is more annoying thing.

PanRe, 2020-09-08:
As i mentioned before, i already figured it out how to handle non-power of two buffer sizes. I did that in my RERO project as well, where i built a ring buffer filled by DMA from USART.
Well we can pospone that surely but as i said then there is this extra copy process necessary since an additional FIFO is needed. If i have time left i prepare a PR for that.
For now, i will wait for @kasjer to finish his tests on the driver and as soon as this is finished i prepare a PR for you to get an audio driver!

kasjer, 2020-09-08:
Hi there, I'm finishing speaker part. It turned out that for this to work correctly I need feedback endpoint and that is taking much time (also it was not clear to me what Windows really expect from audio device in terms of descriptors and it took me some time 
to have it working, while Linux and Mac were happy with what I had originally). FIFO mechanics are not not affecting me, since I made some changes that use just one FIFO for all channels and this change alone drastically reduce CPU time used for FIFO handling.

hathach, 2020-09-09:
> As i mentioned before, i already figured it out how to handle non-power of two buffer sizes. I did that in my RERO project as well, where i built a ring buffer filled by DMA from USART.
> Well we can [...]

Ah great, that would be perfect. Actually after having enough sleep, my brain is functioning back now, and I changed my mind now. Even with the power of 2 limitation, I think we should still opt for a safe implementation (just run into race condition involving mutex and fifo here #507). If you have time, please submit a PR, it is probably easier to keep it as a separated PR (from the ISO).   

PS: Do you think dropping the count could allow us to remove the fifo mutex, I am not entirely sure on this. I guess It may still have issue when the fifo is full and overwritable :thinking: :thinking:

hathach, 2020-09-09:
> Hi there, I'm finishing speaker part. It turned out that for this to work correctly I need feedback endpoint and that is taking much time (also it was not clear to me what Windows really expect from [...]

> For now, i will wait for @kasjer to finish his tests on the driver and as soon as this is finished i prepare a PR for you to get an audio driver!

Superb work, I am looking forward your PR :+1: :+1:

PanRe, 2020-09-09:
My first guess is that you still need the mutexes in case two or more threads try to write into the buffer since you need to protect the write pointer. The same holds for two or more threads reading from the buffer as all of them try to modify the read pointer. What is getting better is that you not need to block the read function while you execute the write function and vice versa (so you could use two separate mutexes for write and read here).For the rest i need a closer look if I have time.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

PanRe, 2020-09-12:
> Hi there, I'm finishing speaker part. It turned out that for this to work correctly I need feedback endpoint and that is taking much time (also it was not clear to me what Windows really expect from [...]

Hi @kasjer! What do you think, when can you bring up a PR for your final tested part?

kasjer, 2020-09-12:
There are some more commits on my branch kasjer/uac2.
It's not all yet, since our work slightly diverged and I have conflicts which are mostly resolved.
You can take a look of what was done.
I also have changes to explicit feedback, that with proper descriptors works with Windows and Linux.
I noticed that @PanRe changes were adding FIFO bypass lately and my work uses FIFO for speaker and microphone so I hope we will not argue here (FIFO code had some minor discrepancies so maybe it was not used much yet).
I expect to push more changes on Monday when I resolve all remaining conflicts, then I will make PR towards @hathach branch so PR for master could be proposed.

kasjer, 2020-09-14:
@PanRe here you are https://github.com/PanRe/tinyusb/pull/2

hathach, 2020-10-08:
@PanRe I think this issue could be closed by now since it is implemented by #516

PanRe, 2020-10-10:
> 
> 
> @PanRe I think this issue could be closed by now since it is implemented by #516

Frankly we are not yet done here ^^. The idea was to convert the EP buffers at least for ISO EPs to FIFOs. The merged unmasked pointer FIFOs are now perfectly suited for this task but it still needs to be implemented (if desired).

hathach, 2020-10-10:
> Frankly we are not yet done here ^^. The idea was to convert the EP buffers at least for ISO EPs to FIFOs. The merged unmasked pointer FIFOs are now perfectly suited for this task but it still needs [...]

Yeah, I think it is is best to be implemented in class driver e.g audio or video. Once it is mature enough we can spin it off like what I plan to do with endpoint stream #509

PanRe, 2021-02-28:
This issue may be closed.

hathach, 2021-04-08:
will closed when #593 is merged

closed 2021-04-15 by hathach