hathach/tinyusb #507: Race condition with cdc auto flush() on transfer completion

hathach, 2020-09-07:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Ubuntu 20.04
 - **Board**   : Feather nRF52840 Express
 - **Firmware**: Arduino with FreeRTOS, original issue is filed here https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/565

Note: The race condition only happen if we use an additional mutex. Semaphore or any other type won't cause the issue. Explanation is below

```
void setup() {

  // put your setup code here, to run once:
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb

  Serial.println("Bluefruit52 USB Serial Test");
  Serial.println("-----------------------------------\n");

  uint32_t t2 = millis() + 5000;
  uint32_t wrsize = 0;
  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  while (millis() < t2) {
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      wrsize += Serial.println("AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb");
      xSemaphoreGive(mutex);
    }
  }
  vSemaphoreDelete(mutex);
  delay(1000);
  Serial.print(wrsize / 5);
  Serial.println("bytes/Sec");
}

void loop() {
}
```

**Screenshots**
[image]

**Describe the bug**
Using above systemview screenshot for reference, the endpoint here is CDC Endpoint In for printing text to host:
- Previously: there is pending transfer on EP
- At event 130: **loop** task call tud_cdc_write() which will acquire the mutex that lock the CDC TX fifo for writing
- 132-136: USB ISR transfer complete 
- 137-139: **usbd** task run and call cdcd_xfer_cb() -> auto flush with tud_cdc_write_flush(), and is blocked when trying to read from tx fifo https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c#L167 (currently held by **loop** at 130). 
- 140: Since usbd is blcoked by mutex, the priority inherit happen -> loop's priority = usbd's priority
- 142-143: **loop** run and finish the write(), release the mutex.
- 144: **loop** continue to run !!! At this time, the **usbd task should be the one running**, since the mutex is released by loop, a context switch should happen, but it is NOT :confused: 
- 146: **loop** call write_flush(), which successfully submit transfer to dcd edpt. loop continue to run until it yield()
- 150-151: just ignore this
- 152: **usbd** task resume and it now read from fifo and try to submit another transfer to an BUSY endpoint !!! DCD will reject this and cause the dropped characters in print() !!!!

**Additional context**

The issue can't happen if at 144 a context switch happens, which it should, and usbd task could complete its write_flush() first, and set the edpt busy. Busy flag will prevent the loop task submitting transfer. **It turns out that  FreeRTOS does not fully implement mutex priority (dis)-inheritance. Priority inherit is not properly reverted when the mutex is released**. Using following example 

[image]

At the end of 3) when **When the LP task gives the mutex back it returns to its original priority.**, in reality, it only go back to its original priority if it doesn't held ANY other mutex, even those the other mutex has nothing to do with the High task. In the above Arduino sketch, the mutex literally does nothing, but still prevent the loop() to go back to its low priority at 144. In fact, the loop() remain in the high priority after the inherit occurs as long as it still hold at least a random mutex 

https://www.freertos.org/FreeRTOS_Support_Forum_Archive/December_2016/freertos_xTaskPriorityDisinherit_when_other_mutexes_are_held_4d6144d1j.html
kernel code https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/tasks.c#L4150

hathach, 2020-09-07:
To make thing even more complicated, here is my drawing of the same process.
[image]

cr1901, 2020-09-08:
May I suggest using LibreOffice Draw to make these sorts of diagrams :)?

hathach, 2020-09-08:
> May I suggest using LibreOffice Draw to make these sorts of diagrams :)?

I will try it next time, maybe I should also try to find an app to make my bad writing look prettier :smile: :smile:

hathach, 2020-09-08:
Update: This race condition can still happen when multiple threads calling `tud_cdc_write_flush()`. I am thinking to add an additional mutex to claim the endpoint first before an API() could make a transfer. Ideally it would be 1 mutex per endpiont, but that seems to be a lot of memory for a edge case. Maybe only one general mutex is enough.

chegewara, 2020-09-08:
> 144: loop continue to run !!! At this time, the usbd task should be the one running, since the mutex is released by loop, a context switch should happen, but it is NOT

Sorry, i dont know nrf, but in general FreeRTOS wont switch task if semaphore/mutex can be claimed (dont have to wait to claim it). Maybe yield() after release mutex can be helpful?

hathach, 2020-09-08:
> Sorry, i dont know nrf, but in general FreeRTOS wont switch task if semaphore/mutex can be claimed (dont have to wait to claim it). Maybe yield() after release mutex can be helpful?

It does switch task when a mutex is released, however due to the  simple implementation of FreeRTOS's priority inheritance, the LP loop() does not go back to its original priority and therefore doesn't switch back when releasing mutex since both task has the same priority now.  

> This schematics about priority inheritance is assuming that both, LP and HP are trying to take the same mutex/semaphore.

Yeah they are trying to get mutex to access the CDC TX fifo .  Yield() every time does fix this particular issue, though I think it isn't the right approach, I am thinking to add usbd_edpt_claim()/release() now, still figuring thing out.

chegewara, 2020-09-08:
There may be another problem that you should look at. Take a look at lines 143, 148, 154. If i understand it correctly then there is no space in Queue probably and queue wait ticks is set to 0. It looks to me there is some code that are trying to push another message to queue before previous got received. Is it possible?
Line 133 looks good, xQueueSendFromISR, but the others not really.

chegewara, 2020-09-08:
> Yield() every time does fix this particular issue, though I think it isn't the right approach

I think it is the correct way to do it. Loop task will get back to its original priority after releasing mutex, but there is not mentioned it will give control to task with higher priority. You have to call delay or yield to let scheduler to switch task.

hathach, 2020-09-09:
> There may be another problem that you should look at. Take a look at lines 143, 148, 154. If i understand it correctly then there is no space in Queue probably and queue wait ticks is set to 0. It [...]

I think it is the default mutex block time https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/include/semphr.h#L40 at those 143 148 154. I remembered freeRTOS implement mutex as queue with depth = 1, item size =0. I could be wrong though.

> Line 133 looks good, xQueueSendFromISR, but the others not really.

Hmm, I don't quite understand your meaning, could you explain it a bit fruther

> I think it is the correct way to do it. Loop task will get back to its original priority after releasing mutex, but there is not mentioned it will give control to task with higher priority. You have to call delay or yield to let scheduler to switch task.

Except it doesn't, since the loop() still held another mutex (the one in the example sketch), the loop() continue to has the high priority of usbd(). Therefore it doesn't switch back since both task has the same prio now. It is due to FreeRTOS opt for simple implementation of prio inheritance.

FreeRTOS kernel code https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/tasks.c#L4150
and it is intended for simple implementation
https://www.freertos.org/FreeRTOS_Support_Forum_Archive/December_2016/freertos_xTaskPriorityDisinherit_when_other_mutexes_are_held_4d6144d1j.html

hathach, 2020-09-09:
I submitted an draft PR for the fix at #508 , if anyone is interested, please give a comment/suggestion there.

Thanos-Love-Ironman, 2020-09-09:
> **Set up**
> [Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible
> 
> * **PC OS**   : Ubuntu 20.04
> * **Board**   : Feather nRF52840 Express
> * **Firmware**: Arduino with FreeRTOS, original issue is filed here [adafruit/Adafruit_nRF52_Arduino#565](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/565)
> 
> Note: The race condition only happen if we use an additional mutex. Semaphore or any other type won't cause the issue. Explanation is below
> 
> ```
> void setup() {
> 
>   // put your setup code here, to run once:
>   Serial.begin(115200);
>   while ( !Serial ) delay(10);   // for nrf52840 with native usb
> 
>   Serial.println("Bluefruit52 USB Serial Test");
>   Serial.println("-----------------------------------\n");
> 
>   uint32_t t2 = millis() + 5000;
>   uint32_t wrsize = 0;
>   SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
>   while (millis() < t2) {
>     if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
>     {
>       wrsize += Serial.println("AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb");
>       xSemaphoreGive(mutex);
>     }
>   }
>   vSemaphoreDelete(mutex);
>   delay(1000);
>   Serial.print(wrsize / 5);
>   Serial.println("bytes/Sec");
> }
> 
> void loop() {
> }
> ```
> 
> **Screenshots**
> [image]
> 
> **Describe the bug**
> Using above systemview screenshot for reference, the endpoint here is CDC Endpoint In for printing text to host:
> 
> * Previously: there is pending transfer on EP
> * At event 130: **loop** task call tud_cdc_write() which will acquire the mutex that lock the CDC TX fifo for writing
> * 132-136: USB ISR transfer complete
> * 137-139: **usbd** task run and call cdcd_xfer_cb() -> auto flush with tud_cdc_write_flush(), and is blocked when trying to read from tx fifo https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c#L167 (currently held by **loop** at 130).
> * 140: Since usbd is blcoked by mutex, the priority inherit happen -> loop's priority = usbd's priority
> * 142-143: **loop** run and finish the write(), release the mutex.
> * 144: **loop** continue to run !!! At this time, the **usbd task should be the one running**, since the mutex is released by loop, a context switch should happen, but it is NOT 😕
> * 146: **loop** call write_flush(), which successfully submit transfer to dcd edpt. loop continue to run until it yield()
> * 150-151: just ignore this
> * 152: **usbd** task resume and it now read from fifo and try to submit another transfer to an BUSY endpoint !!! DCD will reject this and cause the dropped characters in print() !!!!
> 
> **Additional context**
> 
> The issue can't happen if at 144 a context switch happens, (which it should but not), then usbd task could complete its write_flush() first, and set the edpt busy. Busy flag will prevent the loop task submitting transfer. **It turns out that FreeRTOS does not fully implement mutex priority (dis)-inheritance. Priority inherit is not properly reverted when the mutex is released**. Using following example
> 
> [image]
> 
> At the end of 3) when **When the LP task gives the mutex back it returns to its original priority.**. in reality, it only go back to its original priority **if it doesn't held ANY other mutex**, even those the other mutex has nothing to do with the High task. In the above Arduino sketch, the mutex literally does nothing, but still prevent the loop() to go back to its low priority at 144. In fact, the loop() remain in the high priority after the inherit occurs as long as it still hold at least a random mutex
> 
> FreeRTOS kernel code https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/tasks.c#L4150
> and it is intended for simple implementation
> https://www.freertos.org/FreeRTOS_Support_Forum_Archive/December_2016/freertos_xTaskPriorityDisinherit_when_other_mutexes_are_held_4d6144d1j.html

Could you please tell me what's the software you use to analyze USB packets just like the  picture shows.

hathach, 2020-09-09:
> Could you please tell me what's the software you use to analyze USB packets just like the picture shows.

It is segger system view

Thanos-Love-Ironman, 2020-09-09:
> > Could you please tell me what's the software you use to analyze USB packets just like the picture shows.
> 
> It is segger system view

[https://github.com/hathach/tinyusb/issues/503](url)This my issue.
It shows this after cdc port crash.waht does USB  STNC RESET PIPE AND CLEAR STALL means?
[image]

Thanos-Love-Ironman, 2020-09-09:
> > Could you please tell me what's the software you use to analyze USB packets just like the picture shows.
> 
> It is segger system view

Could you please tell me how to use segger system with TinyUSB?It might be the only way to troubleshoot my YinyUSB issue.

chegewara, 2020-09-09:
Here is simplified example i am running on esp32 S2 (also FreeRTOS) to see if task are switching with mutex:
```
int wrsize;
void task(void* p)
{
  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  delay(100);
  String text = String((char*)p);
  while (1) {
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      Serial.println(text);
      xSemaphoreGive(mutex);
    }
    delay(1); // to switch tasks and let setup run
  }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb
  char* task1 = "task1 task1 task1";
  xTaskCreate(task, "task", 4*1024, task1, 21, NULL);

  Serial.println("Bluefruit52 USB Serial Test");
  Serial.println("-----------------------------------\n");

  uint32_t t2 = millis() + 5000;
  uint32_t wrsize = 0;
  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  while (millis() < t2) {
      Serial.printf("%s%d\n", "Setup2: priority = ", uxTaskPriorityGet(NULL));
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      delay(10);
      Serial.printf("%s%d\n", "Setup2: priority = ", uxTaskPriorityGet(NULL));
      xSemaphoreGive(mutex);
    }
      Serial.printf("%s%d\n", "Setup2: priority = ", uxTaskPriorityGet(NULL));
  }
  vSemaphoreDelete(mutex);
  delay(1000);
  Serial.print(wrsize / 5);
  Serial.println("bytes/Sec");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Ask you can see from logs, setup task is never inheriting task priority from other tasks:
```
17:07:03.985 -> Setup1: priority = 1
17:07:03.985 -> task1 task1 task1
17:07:04.018 -> task2 task2 task2task1 task1 task1
17:07:04.018 -> 
17:07:04.018 -> task1 task1 task1
17:07:04.018 -> task2 task2 task2task1 task1 task1
17:07:04.018 -> 
17:07:04.018 -> task1 task1 task1
17:07:04.018 -> task2 task2 task2task1 task1 task1
17:07:04.018 -> 
17:07:04.018 -> Setup2: priority = 1
17:07:04.018 -> task1 task1 task1
17:07:04.018 -> task2 task2 task2task1 task1 task1
17:07:04.018 -> 
17:07:04.018 -> Setup3: priority = 1
17:07:04.018 -> task1 task1 task1
17:07:04.018 -> task2 task2 task2task1 task1 task1
17:07:04.018 -> 
17:07:04.018 -> Setup1: priority = 1
17:07:04.018 -> task1 task1 task1
17:07:04.018 -> task2 task2 task2task1 task1 task1
17:07:04.051 -> 
17:07:04.051 -> task1 task1 task1
17:07:04.051 -> task2 task2 task2task1 task1 task1
17:07:04.051 -> 
17:07:04.051 -> task1 task1 task1
17:07:04.051 -> task2 task2 task2task1 task1 task1
17:07:04.051 -> 
17:07:04.051 -> Setup2: priority = 1
17:07:04.051 -> task1 task1 task1
17:07:04.051 -> task2 task2 task2task1 task1 task1
17:07:04.051 -> 
17:07:04.051 -> Setup3: priority = 1
17:07:04.051 -> task1 task1 task1
17:07:04.051 -> task2 task2 task2task1 task1 task1
```
`Setup2: priority = 1` is always original priority.

chegewara, 2020-09-09:
Now lets consider my original answer, both task should usae the same mutex:
```
int wrsize;
  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();

void task(void* p)
{
//  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  delay(100);
  String text = String((char*)p);
  while (1) {
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      Serial.println(text);
      xSemaphoreGive(mutex);
    }
    delay(1); // to switch tasks and let setup run, can be delay(1)
  }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb
  char* task1 = "task1 task1 task1";
  char* task2 = "task2 task2 task2";
  xTaskCreate(task, "task", 4*1024, task1, 21, NULL);
//  xTaskCreate(task, "task", 4*1024, task2, 10, NULL);

  Serial.println("Bluefruit52 USB Serial Test");
  Serial.println("-----------------------------------\n");

  uint32_t t2 = millis() + 5000;
  uint32_t wrsize = 0;
//  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  while (millis() < t2) {
      Serial.printf("%s%d\n", "Setup1: priority = ", uxTaskPriorityGet(NULL));
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      delay(10);
      Serial.printf("%s%d\n", "Setup2: priority = ", uxTaskPriorityGet(NULL));
      xSemaphoreGive(mutex);
    }
      Serial.printf("%s%d\n", "Setup3: priority = ", uxTaskPriorityGet(NULL));
  }
  delay(1000);
  Serial.print(wrsize / 5);
  Serial.println("bytes/Sec");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Here are logs that proves it is correct way by showing setup is inheriting priority, and HP task is running only after mutex is released:
```
17:12:06.765 -> Setup1: priority = 1
17:12:06.791 -> Setup2: priority = 21
17:12:06.791 -> task1 task1 task1
17:12:06.791 -> Setup3: priority = 1
17:12:06.791 -> Setup1: priority = 1
17:12:06.791 -> Setup2: priority = 21
17:12:06.791 -> task1 task1 task1
17:12:06.791 -> Setup3: priority = 1
17:12:06.791 -> Setup1: priority = 1
17:12:06.791 -> Setup2: priority = 21
17:12:06.791 -> task1 task1 task1
17:12:06.791 -> Setup3: priority = 1
17:12:06.791 -> Setup1: priority = 1
17:12:06.824 -> Setup2: priority = 21
17:12:06.824 -> task1 task1 task1
17:12:06.824 -> Setup3: priority = 1
17:12:06.824 -> Setup1: priority = 1
17:12:06.824 -> Setup2: priority = 21
17:12:06.824 -> task1 task1 task1
17:12:06.824 -> Setup3: priority = 1
17:12:06.824 -> Setup1: priority = 1
17:12:06.824 -> Setup2: priority = 21
```

hathach, 2020-09-09:
@chegewara I am appreciate the time you spent testing out this issue. Though I am bit confused by your 2 example. Can I ask 
1. Did the Serial.printf() in your esp32s2 also use TinyUSB CDC as nrf52840 ?
2. Are you trying to say FreeRTOS implementation is correct, in that case, please visit the link to freeRTOS forum I posted above here is the rtel response. And the code for this function hasn't changed since that anwser. 
```
xTaskPriorityDisinherit when other mutexes are held
Posted by rtel on December 8, 2016
This is a simplification in the priority inheritance implementation done 
in the interest of simplicity.  A full priority inheritance 
implementation is complex due to the number of scenarios (there are many 
combinations of the order in which tasks of various priority can take 
and release semaphores) and thus results in a lot of additional code and 
data space requirements for scenarios that almost never occur – hence we 
opted to implement this simplification.
```

chegewara, 2020-09-09:
1. No, i am not using tinyusb at all in my test code, it was just test of freertos behavior with mutex
2. yes, ive been reading all from link you posted at first place and i am also confused, thats why i prepared this test code

Im not saying that freertos works good or bad, but my understanding of freertos is telling me that all tasks should share the same mutex if you want to block HP by LP task.

Lets imagine situation you have 2 LP tasks and 2 HP tasks.
- task1 and task2 are LP,
- task3 and task4 are HP,
- you want to hold task3 by task1 with mutex and task4 by task2 by mutex,
- if all tasks would use its own mutexes then tasks1 would hold down tasks3 and tasks4 by taking mutex, the same situation with task2,
- when task1 and task3 are using the same mutex, also task2 and task4 are using another mutex, then there is no conflict.
But it may be just my expectations which is different from mutex concept in freertos.

hathach, 2020-09-09:
> 1. No, i am not using tinyusb at all in my test code, it was just test of freertos behavior with mutex

I see, your 2 examples are not the same as the above sketch running on nrf52 with TinyUSB CDC for Serial.printf.

> 2. yes, ive been reading all from link you posted at first place and i am also confused, thats why i prepared this test code

I think you completely misunderstand my point, it probably due to my bad writing. Please re-read my first post again, I will make a short  sumup here again:

Please take a look again at Figure 67 of FreeRTOS explaining the mutex causing the priority inheritance. At step 3 after releasing the mutex that block High task, instead of go back its original low priority as the picture claim, the current FreeRTOS implementation **does not do that** as long as if the LP task still held **any other random/local** mutex which it could acquire anywhere before that chart start.

PS: This issue is about TinyUSB not freeRTOS, I am not in position to judge FreeRTOS as well. They opted for simpler implementation, that's why I don't file a bug issue there. Rather I am trying to find a way to adapt with it. The edpt race exists regardless with another setup as well. I just don't expect to find it out with unexpected behavior from FreeRTOS.

chegewara, 2020-09-09:
Ok, here is quick test that proves your words with that case:
```
Setup3: priority = 1
Setup1: priority = 1
Setup2: priority = 21
task1 task1 task1
Setup3: priority = 21
Setup1: priority = 21
Setup2: priority = 21
task1 task1 task1
Setup3: priority = 21
```
When setup helds 2 mutexes, 1 shared with task priority 10, the other with task priority 21 and 1st mutex is not released, then priority of setup stays at highest inherited level. Only releasing all mutexes will let setup get back to original priority.

```
int wrsize;
  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  SemaphoreHandle_t random_mutex = xSemaphoreCreateMutex();

void task(void* p)
{
//  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  delay(100);
  String text = String((char*)p);
  while (1) {
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      Serial.println(text);
      xSemaphoreGive(mutex);
    }
    delay(1); // to switch tasks and let setup run, can be delay(1)
  }
}

void task0(void* p)
{
//  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  delay(100);
  String text = String((char*)p);
  while (1) {
    if (xSemaphoreTake(random_mutex, portMAX_DELAY) == pdTRUE)
    {
      Serial.println(text);
      xSemaphoreGive(random_mutex);
    }
    delay(1); // to switch tasks and let setup run, can be delay(1)
  }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb
  char* task1 = "task1 task1 task1";
  char* task2 = "task2 task2 task2";
  xTaskCreate(task, "task", 4*1024, task1, 21, NULL);
  xTaskCreate(task0, "task", 4*1024, task2, 10, NULL);

  xSemaphoreTake(random_mutex, portMAX_DELAY);
  
  uint32_t t2 = millis() + 5000;
  uint32_t wrsize = 0;

  while (millis() < t2) {
      Serial.printf("%s%d\n", "Setup1: priority = ", uxTaskPriorityGet(NULL));
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      delay(10);
      Serial.printf("%s%d\n", "Setup2: priority = ", uxTaskPriorityGet(NULL));
      xSemaphoreGive(mutex);
    }
      Serial.printf("%s%d\n", "Setup3: priority = ", uxTaskPriorityGet(NULL));
  }
  delay(1000);
  Serial.print(wrsize / 5);
  Serial.println("bytes/Sec");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

hathach, 2020-09-09:
@chegewara There is no need to post an example here to prove my words at all. I already having an example sketch and the rtel response above already nailed it down. You are probably interested in freeRTOS, but I am only interested in fixing my stack. 

If you got my point now, would you mind if I remove all of our converstation so far on the FreeRTOS, it is a bit distraction with lots of lengthy example. I am worried people will be off-topic from the actual race condition. Thank you.

> In that case maybe using mutex in tinyusb is not correct way? Just thinking out loud.
That is not logical thinking at all, TinyUSB does work with all the task. And have you looked at my PR that fix this issue. !!!

hathach, 2020-09-11:
> In that case maybe using mutex in tinyusb is not correct way? Just thinking out loud.

@chegewara actually, thinking about this, It may make some sense. Maybe TinyUSB could implement osal_mutex as normal semaphore instead :thinking: . Of course, we will fall into the `priority inversion` scenario, though I am not sure between the 2 cases, which is worst hmmm.

chegewara, 2020-09-11:
I am thinking about it and i see 2 potential solutions:
- use binary semaphore instead of mutex, 
> Binary semaphores and mutexes are very similar but have some subtle differences: Mutexes include a priority inheritance mechanism, binary semaphores do not. This makes binary semaphores the better choice for implementing synchronisation (between tasks or between tasks and an interrupt), and mutexes the better choice for implementing simple mutual exclusion.

https://www.freertos.org/Embedded-RTOS-Binary-Semaphores.html
- in tinyusb just before you are trying to claim mutex, store task priority, then right after releasing mutex restore task priority; im not sure if tinyusb allows to access more than 1 task to code that will try to claim mutex, because this wont work in such case (priority will be kept only for last task)

hathach, 2020-09-11:
> * in tinyusb just before you are trying to claim mutex, store task priority, then right after releasing mutex restore task priority; im not sure if tinyusb allows to access more than 1 task to code [...]

this is too much, a usb stack can never fix an RTOS kernel issue. We are better to leave the prio correction to the kernel. Else we will be gonna introduce more bugs.  IMO, the alternative would be implement mutex as binary semaphore on FreeRTOS. Other RTOS such as newt will remain to use mutex. ThoughI am not sure the **priority inversion** vs "being on high prio with random mutex" which scenario is worse.

chegewara, 2020-09-11:
Maybe there is some point of priority inheritance, but i dont see it here now. It is only my personal point of view now, but binary semaphore should be good enough. Maybe @me-no-dev or @igrr has something to say about it.

hathach, 2020-09-14:
Thanks @chegewara for helpful discussion, though I decided to leave it as mutex (and move on with other works). The semaphore vs mutex can be done as follow-up PR if needed. After all the scenario is considered as "almost never occur" by FreeRTOS developers.

PR #508 (2020-09-14): Fix edpt xfer race condition
**Describe the PR**
- fix #507 introduce optional usbd_edpt_claim, usbd_edpt_release which can be ued to gain exclusive access to usbd_edpt_xfer
- With class driver API() that submit transfer to endpoint. Currently `usbd_edpt_busy()` is checked before calling `usbd_edpt_xfer()` e.g

```
uint32_t tud_cdc_n_write_flush (uint8_t itf)
{
  cdcd_interface_t* p_cdc = &_cdcd_itf[itf];

  // skip if previous transfer not complete yet
  TU_VERIFY( !usbd_edpt_busy(TUD_OPT_RHPORT, p_cdc->ep_in), 0 );

  uint16_t count = tu_fifo_read_n(&p_cdc->tx_ff, p_cdc->epin_buf, sizeof(p_c [... cut]
closed 2020-09-14 by hathach