hathach/tinyusb #513: weak symbol relocation error

HiFiPhile, 2020-09-14:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Windows 10
 - **Board**   : STM32L072
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**

Hi, I'm evaluating Segger Embedded Studio with `gcc` and `segger linker` and I got errors with all functions defined as weak while linking.
For example:
```
case DCD_EVENT_RESUME:
        if (tud_resume_cb) tud_resume_cb();
      break;
```
```Relocation R_ARM_ABS32 to weak symbol 'tud_resume_cb' is invalid```

 At first I though it's a linker bug, after discussion with their technical support, I think it's better to modify weak function calls, at least for `gcc arm eabi`.

The `if(weak_symbol)` relocation is not a valid relocation of an undefined weak symbol accord to the AEABI document.  We don't need to do the "if" at all.  Simply call the weak reference and, if it is satisfied with a weak or strong definition, the linker lays down a call to the function.  If the weak reference is not defined, then the linker replaces the call with a "NOP".

This behavior is specified in the AAELF ("ELF for the ARM Architecture") document:
[image]

HiFiPhile, 2020-09-14:
Meanwhile I'm trying to convince Segger to add this support, since all other linkers I've used have no problem.

hathach, 2020-09-14:
Hmm, I also use segger embedded studio for debugging occasionally and didn't have any issues with linking weak function. Have you tried out those project in the example https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_msc/ses

Note: some may be out of date and needs update to compile. 

> This behavior is specified in the AAELF ("ELF for the ARM Architecture") document:

I am aware of the direct use of weak function which expanded to NOP. However, I am not sure how other arch would implement that e.g msp430, or xtensa and/or any future ports. Therefore I am opt to the current `if (weakfunc)`

HiFiPhile, 2020-09-14:
>Hmm, I also use segger embedded studio for debugging occasionally and didn't have any issues with linking weak function.

Do you use segger linker ? I'm using version 5.10a.

It's a demo project about this issue.
[demo.zip](https://github.com/hathach/tinyusb/files/5219079/demo.zip)

hathach, 2020-09-14:
> > Hmm, I also use segger embedded studio for debugging occasionally and didn't have any issues with linking weak function.
> 
> Do you use segger linker ? I'm using version SES 5.10a with segger-ld 3.04.
> 
> It's a demo project about this issue.
> [demo.zip](https://github.com/hathach/tinyusb/files/5219079/demo.zip)

I will try it later on, meanwhile can you compile with the nrf5x project here https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_msc/ses/nrf5x 

It works with my SES 4.52.

HiFiPhile, 2020-09-17:
> meanwhile can you compile with the nrf5x project here

I can compile this project since it's using gcc linker.

I think a solution to ensure compatibility between processors is to add a stub function instead using `if'

For example in `usbd.c`, change 
```
// invoke callback
        if (tud_umount_cb) tud_umount_cb();
```
to
```
TU_ATTR_WEAK void tud_umount_cb(){}
  ...
        if (tud_umount_cb) tud_umount_cb();
```

hathach, 2020-09-24:
> I can compile this project since it's using gcc linker.
>
> I think a solution to ensure compatibility between processors is to add a stub function instead using `if'
>
> For example in `usbd.c`, change [...]

@HiFiPhile Hmm this is a bit complicated than needed, maybe we can just execute the weak function as you mentioned earlier. However I still need to double check that behavior on some of popular platform, and also try to peak to linux code library implementation such as newlib to see if it does the same. If it does, then we probably don't have to overworry.

HiFiPhile, 2020-09-24:
> I still need to double check that behavior on some of popular platform.

Maybe we won't support x86 one day ? Because execute weak function can cause segment fault on x86.

HiFiPhile, 2020-12-22:
It has been fixed in the latest version of SES, with the linker option `--undefined-weak-is-zero`

hathach, 2020-12-22:
thank you for the update,  I guess we could then close this issue by now ?

HiFiPhile, 2020-12-22:
Yes, but maybe put a word somewhere ?

hathach, 2020-12-22:
I guess this issue is good enough, most user could google this out. Or we could point them to this as reference

closed 2020-12-22 by HiFiPhile