hathach/tinyusb #515: SAMD11 HID Examples Send Tons of Random String Descriptors

zlittell, 2020-09-16:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Windows 10
 - **Board**   : SAMD11 XPLAINED
 - **Firmware**: examples/device/hid_composite
                          examples/device/hid_generic_inout

**Describe the bug**
When programming either of these examples to the SAMD11, they constantly send String Descriptors to the host. It Doesn't seem like much of a pattern (it will sometimes send string index 1, then string index 2&3, then just 2, then 1,2,&3 etc). At first I thought this was an issue with creating my project, but it seems the examples do it too. The other examples (nonHID) enumerate correctly swapping descriptors and then just send SOF packets periodically.

**To reproduce**
Steps to reproduce the behavior:
Program either of the HID examples to the SAMD11 XPLAINED

**Screenshots**
This is the HID Example trying to enumerate:
[image]

This is a correct enumeration from Atmel Starts composite HID device example:
[image]

**Log**
Plugged in the EDBG and couldn't get anything on the COM port. Does logging work on the SAMD11?

zlittell, 2020-09-16:
Funny enough it finally quiets down and starts only sending NAKs and SOFs eventually.

Some timed runs of how long it takes:
1)     1minute and 24 seconds 
2)     26 seconds
3)     22 seconds
4)     1 minute 34 seconds
5)     1 minute 12 seconds

The host doesn't send a suspend or anything, it just stops flooding descriptors

hathach, 2020-09-21:
Thanks for filing the issue, are you using an IDE to compile the example, if yes then try to reproduce to see if the issue occur with existing makefile target.   

> Plugged in the EDBG and couldn't get anything on the COM port. Does logging work on the SAMD11?

Yeah, it doesn't work since uart API is not implemented on the board, maybe you could help to add the support for it ? https://github.com/hathach/tinyusb/blob/master/hw/bsp/samd11_xplained/samd11_xplained.c#L126

Since I don't have any samd11 board, I couldn't help testing this out. If you could get the LOG, I will try to analyze it.

zlittell, 2020-09-21:
It happens whether I use an IDE or just compile the examples using make. I will try to get some time to look at the uart stuff. Also I have a SAMD21 xplained board in the drawer and not sure if I can get one of the other examples to run on it.

hathach, 2020-09-23:
> It happens whether I use an IDE or just compile the examples using make. I will try to get some time to look at the uart stuff. Also I have a SAMD21 xplained board in the drawer and not sure if I can get one of the other examples to run on it.

thanks for confirmation, samd21 should work well, since it is widely tested with Adafruit M0 board, though the SAMD21 xplained it not part of supported boards, you may need to add a port for it. SAMD11 mcu is only ported recently, maybe it has some hipccups, since I don't have any boards with samd11, I couldn't test it out at the moment. Please try to get the LOG with your board, it show what is going within the stack and may help to troubleshoot it.  Btw, could you test it with another PC e.g Linux, Mac to see if it is OS specific.

zlittell, 2020-10-03:
I haven't had a chance to test with another OS yet. I have tried it on different hardware but all windows.
Also I spent some time working on the UART today using the hid_composite example and it doesn't fit. I even tried to conserve space using masks to load the registers instead of including any of the HAL/HPL sercom functions. You can test this as well as just including LOG=2 without any of the functionality yet. It compiles fine but fails to program to the device.