hathach/tinyusb #522: 2+ HID Interfaces on a single device

zlittell, 2020-09-30:
**Describe what the question is**
I see the composite example uses a keyboard and mouse on the same interface. When I try to add multiple interfaces I start to get "device not started" issues. The descriptors come in and then it stalls. Is what I am trying to do supported by TinyUSB? If so can someone who has experience with it give me some pointers on what needs to be done to get this type of device rolling.

Most other libraries have you list the report descriptors separately per interface.

Here is a GIST to play with:
https://gist.github.com/zlittell/ff8ca12fdf11228a6f95212dd6a2cb31

hathach, 2020-10-01:
90% of the work for multiple HIDs are done, though it is not very often used by most application. If you are willing to submit pr to add HID multiple interfaces support. I am happy to review. You could take a look at  multiple CDC API for reference 
https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.h#L61

closed 2020-10-01 by hathach

zlittell, 2020-10-01:
working on this now. making good progress, so hopefully all works as planned

hathach, 2020-10-01:
> working on this now. making good progress, so hopefully all works as planned

great to hear that, the most annoying part is we have to add `itf` arguments to all of the callbacks() API, which break current applications. I should have added that in the first place.