hathach/tinyusb #526: When receiving a Get Descriptor request, wValue does not contain lowerbyte.

zlittell, 2020-10-02:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Windows 10
 - **Board**   : SAMD11_xplained
 - **Firmware**: examples/device/hid_multipleinterface

**Describe the bug**
When you read in a Get Descriptor request the high byte of wValue should be 0x22 which it is. The low byte should be the index of the descriptor, but it is always 0x00.

**To reproduce**
Steps to reproduce the behavior:
1. Debug the Multiple Interface HID Demo
2. Set a break point in hid_device.c where tud_hid_descriptor_report_cb is called
3. The lower part of wValue is split out using tu_u16_low(request->wValue) so you can just monitor this variable.

**Screenshots**
[image]
[image]
As you can see in the screenshots I am on interface 0x01 and that should be descriptor index 0x01 as well but wValue is still 0x2200

**Details**
I have traced this back all the way up to where it is read in by the FIFO. The event always has the wrong wValue. I can't figure out how the FIFO would read in a series of bytes and mess up 1 in the middle of the series. Still need to verify is the FIFO actually reads it in wrong, or when it gets shoved into an event something weird happens. Wondering if it might just be an issue with SAMD11 possibly?

Probably low priority unless someone needs it for something else. Just determined the descriptor index based on the interface number, but I didn't want this issue to get lost and forgotten.

hathach, 2020-10-07:
I think you mis-interpreter the request
https://github.com/hathach/tinyusb/blob/440e23c49181687790d9e18e62af501310b62d33/src/common/tusb_types.h#L432

From your capture screenshot the value is received correctly with `value = 0x2200 and index = 0x0001` . As mentioned in the PR, the wIndex is the interface number and since each interface only has 1 hid descriptor, therefore the **desc_index which is lower byte of wValue = 0x00 is correct**. 

I think you mixed up the HID report descriptors, they are not stack up as index 0 and 1 to the host. Host will only consider it as an array if they belong to the SAME interface. Here is the break out, your example 2 hid interfaces. 
- itf 0 has ONLY 1 keyboard report descriptor --> its index is 0
- itf 1 has ONLY 1 mouse report descriptor -> its index is also 0

Host will send request to `itf 0 with index 0` for keyboard, `itf 1 with index 0` for mouse. I hope I explain it clearly enough now.

zlittell, 2020-10-07:
yep looked at the screenshots again and I am not sure what I was thinking on this day. Assumming I just saw the 2201 my brain was expecting and didn't see the 0x00 that came before 0x22 and the one that came after 0x01

closed 2020-10-07 by zlittell

hathach, 2020-10-08:
yeah, that happens when we debugging for too long :)