hathach/tinyusb #537: Race protection for multiple core MCU

hathach, 2020-10-14:
**Is your feature request related to a problem? Please describe.**
Multiple cores MCUs without RTOS can behave like multiple thread RTOS (each core is a thread), therefore we will need to have mutex to prevent race condition e.g
-fifo mutex
- usbd endpoint claim mutex

mbruno-xmos, 2021-08-23:
Related, though perhaps not exactly the same, it would be nice if tud_task() was safe to call from multiple RTOS threads.

I have an application that does USB audio, which requires that the UAC2 driver be pretty responsive with low latency. The same same application also handle vendor requests on endpoint 0 inside tud_vendor_control_xfer_cb(). Responding here can be slow, and can cause tud_vendor_control_xfer_cb() to block for several milliseconds. While it is blocked, the audio endpoint transfers cannot be handled.

I have found that starting two RTOS threads that each call tud_task() sovles this issue. While one is blocked inside tud_vendor_control_xfer_cb(), the other is able to handle the audio endpoint transfers. But I am not sure if this is truly safe and might have several potential race conditions.

hathach, 2021-08-25:
tud_task() is safe to called from RTOS as long as osal mutex is implemented. tud_task() essentially get event from rtos queue to process. This issue is more about MCU with multiple cores running on bare metal (no RTOS) and 2 cores can potentially cause race condition to each other. Btw which RTOS you are running.

Note: current vendor driver isn't thread safe , which is mentioned here.  https://github.com/hathach/tinyusb/issues/900#issuecomment-870261839 Therefore it could be an issue, I will try to bring it up to date later on whenever I could.