hathach/tinyusb #542: Host mode on Adafruit Feather nRF52840 Express

ottobonn, 2020-10-20:
Hi there! I am diving in to TinyUSB because I am interested in using an Adafruit Feather nRF52840 as a USB host for a keyboard. 

I have been playing around with modifying the Adafruit clone of the TinyUSB files to enable host mode, but in switching the `CFG_TUSB_RHPORT0_MODE` to `OP_MODE_HOST` it looks like I will lose the USB serial device for console logs.

I am trying to figure out what configurations this hardware supports and wonder if it's possible to have a USB host on it at all, and if so whether it's possible to have both a device and host. I am open to modifying the hardware if needed, e.g. to add a port.

I am familiar with C programing and happy to work on this but I can't quite figure out how it fits together yet. Any advice would be appreciated!

hathach, 2020-10-22:
nrf52840 hardware controller does not support host mode

closed 2020-10-22 by hathach

ottobonn, 2020-11-01:
Thanks for letting me know. Would you mind sharing which board you use for host development? I am having trouble finding which ones support host mode.

hathach, 2020-11-09:
currently I am using MCB LPC1857 board as main develop platform for host stack.