hathach/tinyusb #549: MSC device cannot be unmounted on Windows

dobairoland, 2020-11-04:
**Set up**

 - **PC OS**   : Windows 10 & Arch Linux 
 - **Board**   : Custom board based on ESP32-S2
 - **Firmware**: Slightly modified version of examples/device/cdc_msc

**Describe the bug**

Linux can unmount the disc without any issue. The device cannot be "ejected" using the same firmware on Windows. (The same computer is used) 

**To reproduce**

Linux:
1. tud_msc_write10_cb() invoked, lba=2, offset=0 - write root directory
2. tud_msc_write10_cb() invoked, lba=0, offset=0 - write boot sector
3. tud_msc_scsi_cb() invoked. bufsize=0, SCSI_CMD_PREVENT_ALLOW_MEDIUM_REMOVAL

Windows:
1. tud_msc_write10_cb() invoked, lba=1, offset=0 - write FAT table
2. tud_msc_scsi_cb() invoked. bufsize=0, SCSI_CMD_PREVENT_ALLOW_MEDIUM_REMOVAL
3. tud_msc_start_stop_cb() invoked, power_condition=0, start=0, load_eject=1

"Safe to remove hardware" message is shown but drive remains attached and can be "ejected" again.

tannewt, 2020-11-05:
I don't think this is up to tinyusb to do. In order to not appear again you need to track the eject and change the readiness of the drive. 

The example doesn't do this because the drive is in RAM and always available. Relevant code is here: https://github.com/hathach/tinyusb/blob/master/examples/device/cdc_msc/src/msc_disk.c#L156

Here is how we do it in CircuitPython: https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/usb/usb_msc_flash.c#L210

hathach, 2020-11-05:
thanks @tannewt for answering, yeah, this is application level code and beyond a usb stack to handle. @dobairoland please analyze the scsi command and its parameter when host issue the msc eject. Then act accordingly to whatever your application want to do.

closed 2020-11-05 by hathach

dobairoland, 2020-11-05:
@tannewt Thank you for the kind help. It is working now.

@hathach I think the example should be ejectable even if the RAM drive is always ready. For example, if I have an USB flash HDD ejected then it is ejected properly and is not showing anymore in "This PC" even though I still have the disk in the USB.

hathach, 2020-11-05:
> @hathach I think the example should be ejectable even if the RAM drive is always ready. For example, if I have an USB flash HDD ejected then it is ejected properly and is not showing anymore in "This PC" even though I still have the disk in the USB. 

It should, would you mind submitting a PR for it ?

dhalbert, 2020-11-05:
We actually use the non-disappearance in CircuitPython so that we can use "Eject" to flush any remaining writes to the disk, but then not have to unplug and plug it back in. There is no other simple way to flush writes. This is particularly important because Windows in the past has not flushed writes to FAT12 drives for up to 90 seconds. Supposedly this has been fixed in newer versions of Windows 10, but I haven't had a chance to test it yet.

I'd like this to have a compile-time switch to disable it, for this reason.

j4cbo, 2020-11-06:
Another way to do this is to un-set the `is_removable` flag in the SCSI INQUIRY response, https://github.com/hathach/tinyusb/blob/a65a0a7996f7ff4cdd50b8d229fb8f9ea01695d1/src/class/msc/msc_device.c#L339