hathach/tinyusb #555: What is the status of the STM32L4 port?

mmalex, 2020-11-11:
A few questions and then details about where I'm coming from
* whats the status of the STM32L4 port?
* do I need to edit the descriptors in the miditest example to make a windows PC recognise it?
* if integrating into a project using ST's HAL libraries, should I configure the USB peripheral first using their routines, or not?
* if not, is the hang in dcd_synopsys.c known about? is there a workaround?
* how is the tinyusb interrupt handler hooked up? I can't see anything in the tinyusb code that would cause dcd_int_handler to actually get called on this platform?

more detail leading to the questions above:
Hi, I realise it's not officially listed in the supported board list, but I noticed a few issues about it, and I would love to help get it running on my STM32L476 project (custom board).
I'm afraid I'm somewhat new to STM32, completely new to tinyusb and usb in general, so I apologise if what I am about to say doesn't make sense.
I'm trying to graft tinyusb midi device mode (using a mostly unmodified copy of the example/miditest code) into a project that otherwise uses ST's CubeIDE / HAL libraries. I've managed to get to a state where the example flashes its LEDs at a rate indicating 'not mounted', and my windows host PC says 'the device malfunctioned and was not recognised'. 
To get there, I initially tried leaving the USB peripheral unconfigured from an ST HAL point of view.
However, when I do this, the dcd_init() routine hangs in the first loop here:
 ```
// Reset core after selecting PHY
  // Wait AHB IDLE, reset then wait until it is cleared
  while ((usb_otg->GRSTCTL & USB_OTG_GRSTCTL_AHBIDL) == 0U) {}
  usb_otg->GRSTCTL |= USB_OTG_GRSTCTL_CSRST;
  while ((usb_otg->GRSTCTL & USB_OTG_GRSTCTL_CSRST) == USB_OTG_GRSTCTL_CSRST) {}

```
if I skip over that first loop in the debugger, the program proceeds however windows doesnt seem to notice any kind of USB activity at all. I also looked at the code and couldn't for the life of me figure out how dcd_int_handler is ever called? dcd_int_enable enables the USB interrupt, but my vector table is in FLASH so it can't be patched at runtime. so how does this ever get called?

my next approach was to enable the boilerplate code generation available in STM32CubeIDE, for USB_OTG_FS peripheral. this powers it up, brings up some awful ST library that seems to enable VDDUSB and stuff, and genraly pokes some registers. it also installs its own giant interrupt handler. This is *without* enabling ST's USB Device library, which I'm trying to avoid. I then run tiny usb code, and it doesnt hang on the loop above, but dcd_int_handler isnt hooked up anywhere either. Windows *does* detect a USB Device, but says it's malfunctioned, I guess this is because the ST code has initialized the peripheral but we haven't replied with any meaningful descriptors - none of the tiny code seems to be running.

I'd happily try to debug this a bit on my board, if anyone can provide me sensible pointers on any or all of these questions. being new to tinyusb Im groping around in the dark a bit. where should I poke first?

thanks for a great project.

dcd_int_handler

mmalex, 2020-11-11:
hello! I was able to make a bit of progress, so I can close this now but in case anyone finds this and wonders the same things as me:
- the STM32L4 port seems to work for midi!
- I discovered that I hadnt copied the hw/bsp/ folder over to my project, and so when searching for the interrupt handler's caller, I failed to find stm32l476disco.c; from that file I learned that it was simply a case of declaring this function in your code, this is weak linked by default to a stub:
`void OTG_FS_IRQHandler(void)
{
  tud_int_handler(0);
}`
I also tried copying the various USB initialization from that file, but didn't get much success. then I went back to the boilerplate created by the STMCube32IDE, at which point I noticed that my USB clock was not 48mhz (oops) - invalidating a lot of my past experiments. I fixed that, and found that the cube-generated boilerplate + correct clock + interrupt handler as above + stock tinyusb midi sample code was enough for my pc to recognise a device! however it failed to initialize the first time; I installed MidiOX at this point on the PC, and the next time I ran (with no changes), the USB device enumerated correctly, and it installed the driver. I dont know if MidiOX was part of that final 'make it work' step, but regardless - I now have a working STM32L476 based board sending Midi messages over USB.
thankyou TinyUSB! hopefully midi in the other direction ends up working also.... I dont believe there's a sample for that?

closed 2020-11-11 by mmalex

hathach, 2020-11-12:
L4 is work in progress, work here and there but also break left and right. you could subscribe to its own issue for update. Though currently I don't have much time to work on L4.