hathach/tinyusb #556: Release name

perigoso, 2020-11-12:
Why do the releases contain the date in the name? this is redundant and creates unnecessary clutter

hathach, 2020-11-12:
I like it that way, what is the reason to not include it ?

perigoso, 2020-11-12:
It clutters the release names, and eventually the archive links, the date is shown on the releases page right under the name.
But if you like it that way there are no major issues I don't think.

hathach, 2020-11-12:
I don't see any issue with link or archive as you mentioned. Can you be specific on the name and link by posting it here

perigoso, 2020-11-12:
You're right it doesn't affect the link, my bad. Again no real issues here, If you have preference for that format.

hathach, 2020-11-12:
no problem, thanks for paying attention to the detail.

closed 2020-11-12 by hathach