hathach/tinyusb #559: tusb_init() returns incorrect value

RomaVis, 2020-11-22:
**Affected versions**
At least 0.7.0 and latest master (589dfdb)

**Describe the bug**
It's a simple coding bug in `tusb_init()` implementation:
```
bool tusb_init(void)
{
  // skip if already initialized
  if (_initialized) return true;

#if TUSB_OPT_HOST_ENABLED
  TU_ASSERT( tuh_init() ); // init host stack
#endif

#if TUSB_OPT_DEVICE_ENABLED
  TU_ASSERT ( tud_init() ); // init device stack
#endif

  _initialized = true;

  return TUSB_ERROR_NONE;
}
```

The function returns **TUSB_ERROR_NONE** (0) while it should return **true** (1).

**Explanation**
Although not documented, from the code we see that if tusb is already initialized and `tusb_init()` is called for the second time, it returns **true**, so **true** indicates successful initialization. However, when initialization is performed successfully for the first time, it returns **TUSB_ERROR_NONE**, which has numeric value **0** that corresponds to **false** in bool.

Consider the following code:
```
if(!tusb_init()) {
  abort();
}
```

Expected behavior: `abort()` is not executed upon successful initialization.
Actual behavior: `abort()` is executed upon successful initialization.

**Proposed solution**

The obvious one is to replace `return TUSB_ERROR_NONE;` with `return true;`. However, this may break existing code that uses tusb, if someone has already implemented workaround for erroneous `tusb_init()` behavior.

P.S.
I don't fully understand the idea behind returning error code / status from this function if it uses `TU_ASSERT` to handle errors... Even if we fix it to return `true` on correct initialization, it will never be able to return `false` because of those assertions...

hathach, 2020-11-23:
> I don't fully understand the idea behind returning error code / status from this function if it uses TU_ASSERT to handle errors... Even if we fix it to return true on correct initialization, it will never be able to return false because of those assertions...

Ah right, thank you for openning the issue, this is left over from legacy code. will fix it right away.

closed 2020-11-23 by hathach

RomaVis, 2020-11-23:
@hathach That was fast. Thank you! :+1:

hathach, 2020-11-23:
I am the one to say thanks, normally I couldn't be that fast, but thankfully this one is easy enough to fix.

PR #560 (2020-11-23): have tusb_init() return true instead of TUSB_ERROR_NONE
**Describe the PR**
fix #559