hathach/tinyusb #567: Webusb example on ESP32-S2

xtrinch, 2020-12-11:
Noob question here:
Will the webusb example work with e.g. ESP32-S2-Saola-1, even though the esp-idf does not support USB host functionality? Or is the USB host functionality not required for webusb?

ladyada, 2020-12-11:
you dont need USB host for webusb. however, we kinda recommend going with web serial instead of web usb because web usb is much harder to use.
https://codelabs.developers.google.com/codelabs/web-serial/?hl=ro#6

xtrinch, 2020-12-11:
I'm wondering then, what makes the S2 a candidate for tinyusb support but not the good old esp32?
The ESP32-S2-Saola-1 board's specifications say that it includes a USB to UART bridge, is that not a problem?

chegewara, 2020-12-11:
webusb, USB serial, USB hid etc are using native USB device, which old esp32 does not have. CP210x on saola board is not used in this case, you may use it only for serial logs or to flash, but all USB device implementations work on native USB which is using pins 19 and 20 on S2.

ladyada, 2020-12-11:
you *can* use webusb with the S2....but we really recommend webserial for web-to-device interaction as it is so much easier to do *and* you can use your old esp32 :)

closed 2020-12-11 by hathach