hathach/tinyusb #570: IAR internal error?

czdpd, 2020-12-17:
Hi,
I try to bring tinyUsb to my project using IAR EWARM(8.5). I added the source files, copied tusb_config.h and usb_descriptor.c from one of the examples. I expect to get a lot compiler errors and I'll work from there. But instead I got IAR internal errors like this:

Tool Internal Error: 
Internal Error: [Front end]: assertion failed at: "layout.c", line 2909 in set_field_size_and_offset 

I wonder if I missed anything, like compiler specific configuration etc? It appears this only happens to these files: tusb.c, usbd.c, usbd_control.c, msc_device.c.

Any idea? Thanks!

HiFiPhile, 2020-12-17:
Hi,

That's quite strange.... Which class driver are you using ?

czdpd, 2020-12-18:
I added cdc, hid, msc sources to my project. The tusb_config.h has CFG_TUD_CDC and CFG_TUD_MSC set to 1 so I guess they're enabled.

HiFiPhile, 2020-12-18:
Could you upload your project ? I can't reproduce on 8.50.6 and 8.50.9

czdpd, 2020-12-18:
Here's my test project. I used ST's example project for the Nucleo STM32L452 board. 
Tinyusb source should be under the directory 'tinyusb'.
IAR project file is at: test_proj\STM32Cube_FW_L4_V1.13.0\Projects\NUCLEO-L452RE\Templates_LL\EWARM

[test_proj.zip](https://github.com/hathach/tinyusb/files/5718370/test_proj.zip)

HiFiPhile, 2020-12-20:
It should be fixed by #573

hathach, 2020-12-23:
closed by #573

closed 2020-12-23 by hathach