hathach/tinyusb #576: Incoming MIDI SysEx packets are mangled

henrix, 2020-12-24:
**Set up**
 - **PC OS**   : macOS 11.1 (Big Sur) 
 - **Board**   : STM32F746-Discovery board
 - **Firmware**: examples/device/midi_test

**Describe the bug**
When sending a MIDI SysEx packet from host with more than 3 bytes, mangled byte array with wrong number of bytes is received. Sending a MIDI SysEx packet with more than 3 bytes works fine though. MIDI SysEx is sent from Web MIDI application.

**To reproduce**
Example code for loopback:
```
uint32_t num_bytes = 0;
uint8_t buf[64] = {0};
if ((num_bytes = tud_midi_available())){
    tud_midi_read(buf, num_bytes);
    tud_midi_write(0, buf, num_bytes);
}
```

**Log**
Example SysEx packet which is sent from Host: `{0xF0, 0x0, 0x1, 0xF7}`.
Packet received at host via loopback: `{0xF0, 0x0, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0xF7}`

prplz, 2021-01-31:
I was having similar issues and I discovered it to be a problem with sysex sending, which I fixed in #612. Does this also fix your problem?

henrix, 2021-02-09:
Sending SysEx works fine already.
As a workaround for receiving SysEx, I'm using the recommendation from #618.

closed 2021-02-09 by henrix