hathach/tinyusb #591: Can't get hid_composite_freertos to compile

nell-du, 2021-01-15:
**Set up** 
 - **PC OS**   : 5.7.19-gentoo
 - **Board**   : esp32s2_saola_1
 - **Firmware**: ../examples/device/hid_composite_freertos

 **Describe the bug**

**_Toolchain setup_**
- I have cloned and updated the **esp-idf** and have checked out the release/v4.2 branch
- I have cloned and updated the **arduino-esp32** library according to [this](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md)
- I have checkout **idf-release/v4.2** like mentioned [here](https://learn.adafruit.com/adafruit-metro-esp32-s2/arduino-ide-setup)
- I have cloned the master branch of tinyusb to a new folder under esp-idf therefore now I have 2 tinyusbs under `~/esp/esp-idf/components/` and they are
```
drwxr-xr-x  4 nelld nelld 4.0K Jan 12 12:04 tinyusb
drwxr-xr-x  3 nelld nelld 4.0K Jan 14 15:14 tinyusb-0.7.0
```
**_To reproduce_**
I went to folder `~/esp/esp-idf/components/tinyusb-0.7.0/tinyusb/examples/device/hid_composite_freertos` to compile the example by calling `make BOARD=esp32s2_saola_1 all` and it gave me 'soc_pins.h' and 'esp_rom_gpio.h' missing. I searched in my $HOME directory and turned out that the files are under `~/Arduino/hardware/espressif/esp32/tools/sdk/esp32s2/include/esp_rom/include` and `~/Arduino/hardware/espressif/esp32/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/`. I would assume that those files at least resides in somewhere in the esp-idf folder but apparently not which is okay I guess I can fix that. 

I then added the above directories to the [CMakeLists.txt](https://github.com/hathach/tinyusb/files/5818213/CMakeLists.txt) under folder `~/esp/esp-idf/components/tinyusb-0.7.0/tinyusb/examples/device/hid_composite_freertos/src'. What it gives me now is that the source files that contain the actual functions are missing. 
```
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp32s2_saola_1/libesp32s2_saola_1.a(esp32s2_saola_1.c.obj):(.literal.configure_pins+0x8): undefined reference to `esp_rom_gpio_pad_select_gpio'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp32s2_saola_1/libesp32s2_saola_1.a(esp32s2_saola_1.c.obj):(.literal.configure_pins+0xc): undefined reference to `esp_rom_gpio_connect_out_signal'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp32s2_saola_1/libesp32s2_saola_1.a(esp32s2_saola_1.c.obj):(.literal.configure_pins+0x10): undefined reference to `esp_rom_gpio_connect_in_signal'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp32s2_saola_1/libesp32s2_saola_1.a(esp32s2_saola_1.c.obj):(.literal.configure_pins+0x14): undefined reference to `esp_rom_gpio_pad_unhold'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/esp32s2_saola_1/libesp32s2_saola_1.a(esp32s2_saola_1.c.obj): in function `configure_pins':
/home/nelld/esp/esp-idf/components/tinyusb-0.7.0/tinyusb/hw/bsp/esp32s2_saola_1/esp32s2_saola_1.c:89: undefined reference to `esp_rom_gpio_pad_select_gpio'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/nelld/esp/esp-idf/components/tinyusb-0.7.0/tinyusb/hw/bsp/esp32s2_saola_1/esp32s2_saola_1.c:91: undefined reference to `esp_rom_gpio_connect_out_signal'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/nelld/esp/esp-idf/components/tinyusb-0.7.0/tinyusb/hw/bsp/esp32s2_saola_1/esp32s2_saola_1.c:93: undefined reference to `esp_rom_gpio_connect_in_signal'
/home/nelld/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/nelld/esp/esp-idf/components/tinyusb-0.7.0/tinyusb/hw/bsp/esp32s2_saola_1/esp32s2_saola_1.c:98: undefined reference to `esp_rom_gpio_pad_unhold'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [../../rules.mk:12: all] Error 2
```
I had a feeling that this could be a toolchain setup issue maybe I messed something up during the process of linking the xtensa-esp32s2-elf to ~/Arduino/hardware/espressif/esp32/tools/ cuz it seems like even though it says version 4.2 supports esp32s2 when you cloned the repository of arduino-esp32 it doesn't actually have the tools for esp32s2. 

Please if you guys could shine some light on this it would be highly appreciated. I would just like to configure the esp32s2 as a USB keyboard meanwhile uploading data to the cloud when there is data coming in from serial.

hathach, 2021-01-15:
can you clone this repo (tip of master) and compile the example with master of esp-idf. IDF changes structure a lot, and tinyusb copy can get out of sync.  https://github.com/hathach/tinyusb/tree/master/examples/device/hid_composite_freertos

chegewara, 2021-01-15:
I can confirm that libray works just fine with esp-idf, arduino-esp32 and arduino as component. There may be small problem to prepare arduino as component project, but its not related to this library.

nell-du, 2021-01-19:
> can you clone this repo (tip of master) separated and compile the example with master of esp-idf. IDF changes structure a lot, and tinyusb copy can get out of sync. https://github.com/hathach/tinyusb/tree/master/examples/device/hid_composite_freertos

Thanks for your answer. I pulled the latest ESP-IDF again and checked out master (now on tip of master). Updated python modules. Checked out master of tinyusb (now on tip of master). The code compiles now. 

Another questions though, I would actually like to NOT use freertos but the `hid_composite` example can't be compiled without a CMakeList.txt. I was wondering do you have one that I can grab right off the bat or what's the plan for that example?

Thank you again for your help.

hathach, 2021-01-19:
I don't think you could run esp32s2 without freeRTOS. If you want, just copy CMake and modify it to do so.

closed 2021-01-19 by hathach