hathach/tinyusb #592: USB communication stops working after handling a vendor control request on the ESP32s2

robojan, 2021-01-16:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Ubuntu 20.04
 - **Board**   :  ESP32-S2-Saola-1
 - **Firmware**: custom firmware with vendor control requests. I'm using the ESP-IDF, but I am not using the included tinyusb. Instead i have cloned this repo as a submodule.

**Describe the bug**
USB communication fails to work after an vendor control request has been handled. 

**To reproduce**
On the  ESP implement the following vendor control xfer callback.
```c++
bool tud_vendor_control_xfer_cb(uint8_t  rhport, uint8_t stage, tusb_control_request_t const *request)
{
    static uint8_t testBuffer[32] = {};
    switch(request->bRequest)
    {
    case 0:
        return tud_control_xfer(rhport, request, (void *)testBuffer,
                                32);
    default:
        return false;
    }
}
```

On the pc run the following test program.
```c++
#include <libusb-1.0/libusb.h>
#include <cstdio>
#include <cassert>
#include <array>
#include <cstdlib>
#include <cstring>

void triggerBug(libusb_device_handle *devHandle)
{
    std::array<uint8_t, 8> buffer;

    // Just read some usb descriptor to see if the usb communication works
    int result = libusb_get_string_descriptor(devHandle, 1, 0x403, buffer.data(), 5);
    if(result < 0) {
        printf("Error reading descriptor: %s\n", libusb_strerror((libusb_error)result));
    } else {
        printf("Success reading descriptor\n");
    }

    // Just read some usb descriptor to see if the usb communication works
    result = libusb_get_string_descriptor(devHandle, 1, 0x403, buffer.data(), 5);
    if(result < 0) {
        printf("Error reading descriptor: %s\n", libusb_strerror((libusb_error)result));
    } else {
        printf("Success reading descriptor\n");
    }

    // Do the transfer that crashes the usb on the ESP side
    result = libusb_control_transfer(devHandle, LIBUSB_ENDPOINT_IN | LIBUSB_REQUEST_TYPE_VENDOR, 
        0, 0, 0, buffer.data(), 8, 10000);
    if(result < 0) {
        printf("Error reading descriptor: %s\n", libusb_strerror((libusb_error)result));
    } else {
        printf("Success reading vendor descriptor\n");
    }

    // Just read some usb descriptor to see if the usb communication works
    result = libusb_get_string_descriptor(devHandle, 1, 0x403, buffer.data(), 5);
    if(result < 0) {
        printf("Error reading descriptor: %s\n", libusb_strerror((libusb_error)result));
    } else {
        printf("Success reading descriptor\n");
    }

    // Just read some usb descriptor to see if the usb communication works
    result = libusb_get_string_descriptor(devHandle, 1, 0x403, buffer.data(), 5);
    if(result < 0) {
        printf("Error reading descriptor: %s\n", libusb_strerror((libusb_error)result));
    } else {
        printf("Success reading descriptor\n");
    }
}

int main(int argc, const char **argv)
{
    int result;

    result = libusb_init(nullptr);
    assert(result >= 0);

    libusb_device_handle *devHandle;
    devHandle = libusb_open_device_with_vid_pid(nullptr, 0x303a, 0x4080);
    if(!devHandle) {
        printf("Could not open device\n");
        return -1;
    }

    triggerBug(devHandle);

    libusb_close(devHandle);

    return 0;
}
```
The test program will print 
```
Success reading descriptor
Success reading descriptor
Success reading vendor descriptor
Error reading descriptor: Overflow
Error reading descriptor: Pipe error
```

**Log**
I had some problems with getting the log to work on the ESP. So i changed some of the ESP_LOGV to ESP_LOGI in the dcd_esp32s2.c file. 
```
I (33340) TUSB:DCD: dcd_int_handler - rx!
I (33340) TUSB:DCD: dcd_int_handler - rx!
I (33340) TUSB:DCD: dcd_int_handler - OUT endpoint!

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes
I (33360) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 5
I (33360) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33360) TUSB:DCD: TUSB IRQ - EP IN 0
I (33360) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33360) TUSB:DCD: USB - transmit_packet
I (33380) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33380) TUSB:DCD: TUSB IRQ - EP IN 0
I (33380) TUSB:DCD: TUSB IRQ - IN XFER complete!
I (33380) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33380) TUSB:DCD: USB - transmit_packet
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with zlp Status
I (33420) TUSB:DCD: Transfer <-> EP0, HOST->DEV (out), pkgs: 1, bytes: 0
I (33420) TUSB:DCD: dcd_int_handler - rx!
I (33420) TUSB:DCD: USB - receive_packet
I (33420) TUSB:DCD: dcd_int_handler - rx!
I (33420) TUSB:DCD: dcd_int_handler - rx!
I (33420) TUSB:DCD: dcd_int_handler - OUT endpoint!
I (33420) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
I (33420) TUSB:DCD: dcd_int_handler - rx!
I (33420) TUSB:DCD: dcd_int_handler - OUT endpoint!
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes
I (33480) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 5
I (33490) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33490) TUSB:DCD: TUSB IRQ - EP IN 0
I (33490) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33490) TUSB:DCD: USB - transmit_packet
I (33510) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33510) TUSB:DCD: TUSB IRQ - EP IN 0
I (33510) TUSB:DCD: TUSB IRQ - IN XFER complete!
I (33510) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33510) TUSB:DCD: USB - transmit_packet
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with zlp Status
I (33540) TUSB:DCD: Transfer <-> EP0, HOST->DEV (out), pkgs: 1, bytes: 0
I (33550) TUSB:DCD: dcd_int_handler - rx!
I (33550) TUSB:DCD: USB - receive_packet
I (33550) TUSB:DCD: dcd_int_handler - rx!
I (33550) TUSB:DCD: dcd_int_handler - rx!
I (33550) TUSB:DCD: dcd_int_handler - OUT endpoint!
I (33550) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
I (33550) TUSB:DCD: dcd_int_handler - rx!
I (33550) TUSB:DCD: dcd_int_handler - OUT endpoint!
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received C0 00 00 00 00 00 08 00 
  Queue EP 80 with 8 bytes
I (33600) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 8
I (33600) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33600) TUSB:DCD: TUSB IRQ - EP IN 0
I (33600) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33600) TUSB:DCD: USB - transmit_packet
I (33620) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33620) TUSB:DCD: TUSB IRQ - EP IN 0
I (33620) TUSB:DCD: TUSB IRQ - IN XFER complete!
I (33620) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33620) TUSB:DCD: USB - transmit_packet
USBD Xfer Complete on EP 80 with 8 bytes
  VENDOR control complete
  Queue EP 80 with 8 bytes
I (33660) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 8
I (33670) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33670) TUSB:DCD: TUSB IRQ - EP IN 0
I (33670) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33670) TUSB:DCD: USB - transmit_packet
  Queue EP 00 with zlp Status
I (33690) TUSB:DCD: Transfer <-> EP0, HOST->DEV (out), pkgs: 1, bytes: 0
I (33700) TUSB:DCD: dcd_int_handler - rx!
I (33700) TUSB:DCD: USB - receive_packet
I (33700) TUSB:DCD: dcd_int_handler - rx!
I (33700) TUSB:DCD: dcd_int_handler - rx!
I (33700) TUSB:DCD: dcd_int_handler - OUT endpoint!
I (33700) TUSB:DCD: TUSB IRQ - EP OUT - XFER complete (single packet)
I (33700) TUSB:DCD: dcd_int_handler - rx!
I (33700) TUSB:DCD: dcd_int_handler - OUT endpoint!
USBD Xfer Complete on EP 00 with 0 bytes
  Queue EP 80 with 8 bytes
I (33740) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 8
I (33750) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33750) TUSB:DCD: TUSB IRQ - EP IN 0
I (33750) TUSB:DCD: TUSB IRQ - IN XFER complete!
I (33750) TUSB:DCD: TUSB IRQ - IN XFER FIFO empty!
I (33750) TUSB:DCD: USB - transmit_packet
I (33750) TUSB:DCD: dcd_int_handler - rx!
I (33750) TUSB:DCD: dcd_int_handler - rx!
I (33750) TUSB:DCD: dcd_int_handler - OUT endpoint!

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes
I (33800) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 5
I (33810) TUSB:DCD: dcd_int_handler - IN endpoint!
I (33810) TUSB:DCD: TUSB IRQ - EP IN 0
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with zlp Status
I (33820) TUSB:DCD: Transfer <-> EP0, HOST->DEV (out), pkgs: 1, bytes: 0

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes
I (33840) TUSB:DCD: Transfer <-> EP0, USB0.HOST (in), pkgs: 1, bytes: 5
```

hathach, 2021-02-14:
Hi, would you mind trying it again with latest from master branch. There is a recent pr that may fix this issue #646

robojan, 2021-02-14:
Hi, I just tried it again with the latest master branch. commit 09868434cd9f53394350cce682333ada97f796c3. 
It still does not work, but now it seems that the asserts are enabled. 

```
USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes ... OK
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes ... OK
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received C0 00 00 00 00 00 08 00 
  Queue EP 80 with 8 bytes ... OK
USBD Xfer Complete on EP 80 with 8 bytes
  VENDOR control complete
  Queue EP 80 with 8 bytes ... OK
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes
  Queue EP 80 with 8 bytes ... usbd_edpt_xfer 1210: assert failed
tud_control_xfer 121: assert failed

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes ... OK
USBD Xfer Complete on EP 80 with 5 bytes
  Queue EP 00 with 0 bytes ... OK

USBD Setup Received 80 06 01 03 03 04 05 00 
  Get Descriptor String[1]
  Queue EP 80 with 5 bytes ... OK
```

It seems it tries to send data on the endpoint while it is busy.
If you are curious this is the project that generates the errors: https://gitlab.com/robojan1/led-cube/-/tree/develop/firmware
I'm trying to implement the microsoft os descriptors for winUSB support. In the past i did something similar on the i.MX RT series from NXP and there were no problems on that chip.

robojan, 2021-02-14:
Ah i figured it out. At some point the tud_vendor_control_xfer_cb has been updated to also be called when the transfer is completed. 

It works properly now after adding this to the callback.
```
    // nothing to with DATA & ACK stage
    if(stage != CONTROL_STAGE_SETUP)
        return true;
```

hathach, 2021-02-15:
Thanks for the update. This issue is resolves

closed 2021-02-15 by hathach