hathach/tinyusb #607: RP2040 hid_generic_inout does not work

hathach, 2021-01-26:
**Set up**
 - **PC OS**   : Ubuntu 20.04 
 - **Board**   : RPI Pico
 - **Firmware**: examples/device/hid_generic_inout

**Describe the bug**
Device doesn't seem to response to hid_test.py script 

**To reproduce**
Run `python3 hid_test.py` in the example folder

```
python3 hid_test.py 
Openning HID device with VID = 0xCAFE
{'path': b'/dev/hidraw9', 'vendor_id': 51966, 'product_id': 16388, 'serial_number': '123456', 'release_number': 256, 'manufacturer_string': 'TinyUSB', 'product_string': 'TinyUSB Device', 'usage_page': 45744, 'usage': 475, 'interface_number': 0}
Send text to HID Device : thach
Traceback (most recent call last):
  File "hid_test.py", line 15, in <module>
    dev.write(str_out)
  File "/home/hathach/.local/lib/python3.8/site-packages/hid/__init__.py", line 155, in write
    return self.__hidcall(hidapi.hid_write, self.__dev, data, len(data))
  File "/home/hathach/.local/lib/python3.8/site-packages/hid/__init__.py", line 146, in __hidcall
    raise HIDException(err)
hid.HIDException: None
```

hathach, 2021-01-27:
update: further testing show that RP2040 does work when attaching directly to PC, it doesn't seem to work when attaching via an external hub.

majbthrd, 2021-01-31:
@hathach, if you add an “#error” to ./src/portable/raspberrypi/rp2040/dcd_rp2040.c at about line 328 inside the #if TUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX block, and then attempt a compile, it doesn’t report the error.  The “rp2040_usb_device_enumeration_fix” appears to be specifically for a problem with (some) USB hubs.

hathach, 2021-02-01:
> @hathach, if you add an “#error” to ./src/portable/raspberrypi/rp2040/dcd_rp2040.c at about line 328 inside the #if TUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX block, and then attempt a compile, it [...]

Ah, you are spot-on, seem like I did missed the macro, it probably only required for this chip version and will be fixed in future silicon, finding a way to insert the macro.

hathach, 2021-02-01:
update: I have added the enumeration fix, but it still doesn't solve the problem with hub. And it does make sense though, the fix only help rp2040 to exit bus reset state (RP2040-E5), in this case the device already enumerated the hid_test.py can successfully get its config and hid descriptor. It is probably something else, but it is nice to catch the missing enum fix :)

hathach, 2021-02-19:
issue still exist with https://github.com/hathach/tinyusb/commit/9067926a70633de448c67126e81e3ae107634405