hathach/tinyusb #608: RP2040 net_lwip_webserver does not work

hathach, 2021-01-26:
**Set up**
 - **PC OS**   : Ubuntu 20.04 
 - **Board**   : RPI Pico
 - **Firmware**: examples/device/hid_generic_inout

**Describe the bug**
cannot access the on-board webserver at 192.168.7.1 . The device seems to enumerate just fine. Further troubleshooting is needed

majbthrd, 2021-01-31:
Hmm, for what it is worth, net_lwip_webserver for the raspberry_pi_pico target works on all PCs (Linux and Windows) that I have locally.

hathach, 2021-02-01:
> Hmm, for what it is worth, net_lwip_webserver for the raspberry_pi_pico target works on all PCs (Linux and Windows) that I have locally. (Ubuntu 20.04.1, Linux Mint 19, Linux Mint 17, Windows 7)

Ah thanks for testing it out, indeed, somehow my PC network configure  only allow 1 ethernet connection at a time and I have to manual toggle the connection. Work fine when selecting the correct USB conneciton

[image]

closed 2021-02-01 by hathach