hathach/tinyusb #609: RP2040 webusb_serial example does not work

hathach, 2021-01-26:
**Set up**
**Set up**
 - **PC OS**   : Ubuntu 20.04 
 - **Board**   : RPI Pico
 - **Firmware**: examples/device/webusb_serial

**Describe the bug**
Device enumerated, landing URL appear on chrome, could connect via chrome. However, there seem to be issue when using along side with cdc serial. Further troubleshooting is needed.

**To reproduce**
1. Click on landing page, click connect
2. try to open terminal with cdc serial
3. RP2040 stop responding

gniezen, 2021-01-27:
I was able to reproduce this issue on Elementary OS (Ubuntu derivative). Windows 10 seems to work fine.

chegewara, 2021-01-27:
I dont know if this is usefull somehow, but webusb (and maybe other hid example) is causing that ubuntu usb driver stop responding/discovering devices at some point:
```
[ 4627.697667] usb 3-1: new full-speed USB device number 13 using xhci_hcd
[ 4627.875037] usb 3-1: New USB device found, idVendor=cafe, idProduct=4011, bcdDevice= 1.00
[ 4627.875041] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4627.875044] usb 3-1: Product: TinyUSB Device
[ 4627.875046] usb 3-1: Manufacturer: TinyUSB
[ 4627.875047] usb 3-1: SerialNumber: 123456
[ 4627.924334] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
[ 4627.928101] usbcore: registered new interface driver cdc_acm
[ 4627.928101] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[ 4697.455999] usb 3-1: USB disconnect, device number 13
[ 4702.852548] xhci_hcd 0000:0a:00.3: xHCI host not responding to stop endpoint command.
[ 4702.852552] xhci_hcd 0000:0a:00.3: USBSTS:
[ 4702.868560] xhci_hcd 0000:0a:00.3: Host halt failed, -110
[ 4702.868563] xhci_hcd 0000:0a:00.3: xHCI host controller not responding, assume dead
[ 4702.868578] xhci_hcd 0000:0a:00.3: HC died; cleaning up
[ 4702.870287] cdc_acm 3-1:1.0: failed to set dtr/rts
[ 4702.871002] usb 3-2: USB disconnect, device number 3
[ 4702.871196] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 4702.871324] cp210x 3-2:1.0: device disconnected
[ 4702.871518] usb 3-3: USB disconnect, device number 4
```
From now connecting any device to USB is not detected.

chegewara, 2021-01-27:
@hathach an observation:
```
Interface Number: 1
		Name: cdc_acm
		Alternate Number: 0
		Class: 0a(data ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 02
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

			Endpoint Address: 82
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

	Interface Number: 2
		Name: usbfs
		Alternate Number: 0
		Class: ff(vend.) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 03
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

			Endpoint Address: 83
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms
```
Interval is 0ms, is it possible its causing driver stuck?
I am using stock examples from tinyusb repo.

hathach, 2021-01-29:
yeah, I saw the same panic as well, looking closely at the driver, though I guess it is due to the write to already-received control endpoint.

ipopov, 2021-02-09:
I ran into the same problem and was able to diagnose it fully.

The problem is that when receiving a "standard" CLEAR_FEATURE request, usbd.c calls into the _class_ driver:
https://github.com/hathach/tinyusb/blob/a96ee8f1d850bbd645040e0f5d133e6c1c93a3a4/src/device/usbd.c#L757. (The comment there says that the class *must not* call tud_control_status.)

Unfortunately the "class driver", i.e. the webusb example's main function, intereprets this as a "class" request with bRequest==1 (i.e. VENDOR_REQUEST_WEBUSB), rather than as a "endpoint" request with bRequest==CLEAR_FEATURE,
https://github.com/hathach/tinyusb/blob/a96ee8f1d850bbd645040e0f5d133e6c1c93a3a4/examples/device/webusb_serial/src/main.c#L156, and sends back a report.

To fix this, the class driver should simply ignore non-class requests.

hathach, 2021-02-09:
can you enable the LOG=2, hook FTDI to GPIO0,1 and paste your log her. The driver works just fine with other mcus therefore it can be a bit more troublesome than mis-representing the request I guess. Though to be honest, I haven't got time to worked on this again.

ipopov, 2021-02-10:
Hey there, Ha,

You're right that the driver working fine on other mcus suggests that the bug is deeper. On the other hand, it's possible that other MCUs ignore a spurious status report, whereas the RP2040 is programmed to panic.

In any case, I did what you requested. The stack trace of the panic is

```
(gdb) bt
#0  _exit (status=status@entry=1) at /home/pi/sdk2/src/rp2_common/pico_runtime/runtime.c:164
#1  0x10001a28 in panic (fmt=0x100088f0 "ep %d %s was already available")
    at /home/pi/sdk2/src/rp2_common/pico_runtime/runtime.c:242
#2  0x10004b7e in _hw_endpoint_buffer_control_update32 (ep=ep@entry=0x20001020 <hw_endpoints+52>,
    and_mask=and_mask@entry=0, or_mask=<optimized out>)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:84
#3  0x10004bc6 in _hw_endpoint_buffer_control_set_value32 (value=<optimized out>,
    ep=0x20001020 <hw_endpoints+52>)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.h:107
#4  _hw_endpoint_start_next_buffer (ep=ep@entry=0x20001020 <hw_endpoints+52>)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:133
#5  0x10004c24 in _hw_endpoint_xfer_start (ep=0x20001020 <hw_endpoints+52>, buffer=0x0,
    total_len=<optimized out>)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:161
#6  0x10004d1c in _hw_endpoint_xfer (ep=<optimized out>, buffer=buffer@entry=0x0,
    total_len=total_len@entry=0, start=start@entry=true)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:272
#7  0x100047fa in hw_endpoint_xfer (ep_addr=ep_addr@entry=128 '\200', buffer=buffer@entry=0x0,
    total_bytes=<optimized out>, start=start@entry=true)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:209
#8  0x10004a64 in dcd_edpt_xfer (rhport=rhport@entry=0 '\000', ep_addr=ep_addr@entry=128 '\200',
    buffer=buffer@entry=0x0, total_bytes=total_bytes@entry=0)
    at /home/pi/sdk2/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c:456
#9  0x10005e82 in _status_stage_xact (request=0x20041fe0, rhport=0 '\000')
    at /home/pi/sdk2/lib/tinyusb/src/device/usbd_control.c:75
#10 tud_control_status (rhport=rhport@entry=0 '\000', request=request@entry=0x20041fe0)
    at /home/pi/sdk2/lib/tinyusb/src/device/usbd_control.c:86
#11 0x1000047c in tud_vendor_control_request_cb (rhport=<optimized out>, request=0x20041fe0)
    at /home/pi/examples2/usb/device/dev_webusb_serial/dev_webusb_serial.c:187
#12 0x10004dfa in invoke_class_control (rhport=rhport@entry=0 '\000',
    driver=0x10008b70 <_usbd_driver+32>, request=request@entry=0x20041fe0)
    at /home/pi/sdk2/lib/tinyusb/src/device/usbd.c:570
#13 0x100059c4 in process_control_request (rhport=<optimized out>, p_request=p_request@entry=0x20041fe0)
    at /home/pi/sdk2/lib/tinyusb/src/device/usbd.c:698
#14 0x10005c2e in tud_task () at /home/pi/sdk2/lib/tinyusb/src/device/usbd.c:494
#15 0x10000596 in main () at /home/pi/examples2/usb/device/dev_webusb_serial/dev_webusb_serial.c:80
```

and the LOG(2) output is

```
USBD init
CDC init
VENDOR init
USBD Bus Reset

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset
  
USBD Setup Received 00 05 20 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
[... 136 lines cut ...]
USBD Setup Received 02 01 00 00 03 00 00 00
  Clear Feature
  Queue EP 80 with zlp Status
  VENDOR control request
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 02 01 00 00 83 00 00 00
  Clear Feature
  Queue EP 80 with zlp Status
  VENDOR control request
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 21 22 01 00 02 00 00 00
  VENDOR control request
  Queue EP 83 with 33 bytes ... OK
  Queue EP 80 with zlp Status

*** PANIC ***

ep 0 in was already available
```

I'll upload a patch expressing my fix.

ipopov, 2021-02-10:
I worked up this patch:
https://github.com/ipopov/tinyusb/commit/fd8add72e83a7df67dbaae9ec00a8a5eedfe26a3

which shows what I mean, and has allowed me to run the WebUSB Serial demo. However, sadly, it is based off of e0aa405d19e35dbf58cf502b8106455c1a3c2a5c, which is the original "RP2040 Support" commit, shipped with the Pico SDK; and that is based on a TinyUSB fork from November 2020.

I'm not really set up currently to build TinyUSB at head, sadly; so I cannot test whether the fix works well at head currently. I might spend some time on this in the coming week, though..

ipopov, 2021-02-10:
I managed to get TinyUSB at head building, and put together a proper pull request: https://github.com/hathach/tinyusb/pull/645. PTAL! :-)

hathach, 2021-02-10:
Thank you for your log, it is very useful. It is indeed issue between how usbd and class driver agree up on. This seems to be added after webusb is already written and on other mcu, the duplicate ZLP has no effect or overwrite by the next setup packet. Therefore it doesn't bring up any attention to fix. Hmm, I think usbd should detect if tud_control_status() is invoked rather than trusting class driver for not calling it, especially it is vendor in user space.

The #645 is spot-on, I am probably too lazy checking the request type. Great finding.

hathach, 2021-02-11:
closed by either #645 or #646 . Thank you @ipopov for troubleshooting the issue.

closed 2021-02-11 by hathach

gniezen, 2021-02-11:
Just tested it from `master` branch on Elementary OS and it worked great. Thanks for looking into this @ipopov , and @hathach for your quick response!