hathach/tinyusb #611: Half of the data accepted by running uac2_headset using esp32s2 is lost

fate100, 2021-01-28:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Ubuntu 18.04
 - **Board**   : esp32s2_kaluga_1
 - **Firmware**: examples/device/uac2_headset_freertos

**Describe the bug**
I use FreeRTOS to port `uac2_headset_freertos`, but when I run it and play audio on the computer, I find that the audio data buffer returned by `tud_audio_rx_done_cb` is only half of the **USB packet capture data**. I tried to compare the two data and found that the audio data buffer returned by `tud_audio_rx_done_cb` will lose a USB isochronous packet (**192 Byte**) whenever it receives a USB isochronous packet (**192 Byte**).

pm-daniel, 2021-01-28:
PC OS : Windows 10 2004
Board : NUCLEO-H743ZI2
Firmware: examples/device/uac2_headset
I've encountered a similar problem using `tud_audio_rx_done_cb` callback to send audio samples using SAI in DMA mode - everything ought to be correct but somehow half of the data is being corrupted, though I'm not sure if it's a bug or my poor understanding of how exactly this library works - both cases are possible.

PanRe, 2021-04-15:
The new driver is out, so you may test your application again. If it does not work, you may look into the dcd driver. There might be an issue with the ISO EPs. You may also test it without RTOS if the problem still remains.

fate100, 2021-04-22:
I used the new driver to try the uac2_headset example again, but when I run it on the computer(windows), the USB Audio driver does not work properly, so I can’t find the TinyUSB Device.
In addition, I also tried to use esp32s2 to run audio_test, but it also failed to start the device.
I suspect there is something wrong with the descriptors, but I am not very clear about the USB_Audio protocol.

hathach, 2021-04-22:
esp32s2 not yet fully support or tested with ISO transfer and new audio driver. It currently only works best with stm32. This will unfortunately stay this way for awhile until I find more time to fully update all other ports.

fate100, 2021-04-23:
I am looking forward to the new audio driver support for esp32s2

noodlefighter, 2021-05-24:
i am trying to run uac2_headset on esp32s2, and UAC device is enumerated successfully (fix by #841).
but It seems cannot receive any data from OUT endpoint.
USB protocol analyzer shows that the OUT-EP endpoint is transmitting data.
any idea?

noodlefighter, 2021-05-26:
@fate100 run uac2_headset on esp32s2 successfully. details: #849

PanRe, 2021-06-12:
Thanks a lot for fixing the example and for porting! The audio driver was under heavy development and we only had our custom boards so we could not re-test all examples. Can this issue be closed?

hathach, 2021-06-12:
yup, let close this, can be opened another one if needed.

closed 2021-06-12 by hathach