hathach/tinyusb #616: hw/bsp/samd21 and hw/bsp/samd51 are missing "sam.h"

charkster, 2021-01-29:
In line 27 of tinyusb/hw/bsp/samd21/family.c the "sam.h" header file is included, but no where to be found.
Also line 27 of tinyusb/hw/bsp/samd51/family.c the "sam.h" header file is included, but no where to be found.

PS OS : Ubuntu 20.04, Raspberry Pi OS
Board: seeeduino xiao
Firmware: examples/device/usbtmc

I am unable to compile the usbtmc example because a header file is missing.
```
/home/pi/tinyusb/hw/bsp/samd21/family.c:27:10: fatal error: sam.h: No such file or directory
 #include "sam.h"
          ^~~~~~~
compilation terminated.
make: *** [../../rules.mk:153: _build/seeeduino_xiao/obj/hw/bsp/samd21/family.o] Error 1
```

Steps to reproduce the behavior:
1. Go to tinyusb/examples/device/usbtmc 
2. Run `make BOARD=seeeduino_xiao all uf2`

hathach, 2021-01-29:
current microchip driver has asf4 in its own submodule https://github.com/hathach/microchip_driver . You need to clone submodule once more level. I will try to see if I could simplify it later on.

```
 git submodule update --init --recursive hw/mcu/microchip
```

charkster, 2021-01-29:
Sorry about that. I missed the step of getting all the submodules. I just updated the submodules for all the hardware and am fine now.

Did the Raspberry Pi guys just recursively added all the libraries again under their own hardware?
/home/pi/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/hw/mcu/microchip
/home/pi/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/hw/mcu/sony/cxd56/spresense-exported-sdk

When I get all the submodules it is twice the size now.

closed 2021-01-29 by charkster