hathach/tinyusb #629: TIVA C SUPPORT

scott-olero, 2021-02-04:
Have you managed to add support for TIVA C ?, im struggling to port ,the MSP430 example to it. Any help would be greatly appreciated 
[dcd_msp430x5xx.zip](https://github.com/hathach/tinyusb/files/5923378/dcd_msp430x5xx.zip)

Thanks.

hathach, 2021-02-04:
I haven't tried to port this to TI Tiva mcu, in fact I haven't even worked with it so far. Though lots of people do, and willing to help you with the port. However, you need to do it in a more collaborating manner by submitting draft PR (zip doesn't help much with discussion). Here is my porting routines, please
1.  fork the repo
2. add mcu low-level driver, bsp for Tiva
3. make sure you got the `board_test` example running with led + button, when pressing button, the LED will blink faster. This example may look irrelevant, but it does help to confirm the clock setup is correct. I normally compare the blinking rate to other board to see if new port blink faster/slower which indicate an issue with clock.
4. Add new `OPT_MCU_*` and new dcd_
5. Make a draft PR, even though it is not working yet. This help us to easily spot issue and have a helpful feedback/discussion as well as testing your latest code

Yeah, It is a lots of works, and it is expected for adding new port. I actually have a TivaC board EK-TM4C123GXL and could help with testing this out.

aspdigital, 2021-08-09:
I have both the TM4C1294 board and the MSP432E board, and I'm happy to help with the port. The chips use the Synopsys USB module. They support High Speed USB with an external ULPI PHY.

hathach, 2021-08-10:
Thanks for your willingness, there is an issue for generalizing synopsys driver to make it easier to integrate to new port #382, however, I haven't got time to do any progress yet for you to test with.

hathach, 2022-04-20:
tiva c support is implemented with #1183

closed 2022-04-20 by hathach