hathach/tinyusb #635: Cannot get Audio to write out at full data sample rate

ndinsmore, 2021-02-06:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : osx 10.15 
 - **Board**   : pi pico
 - **Firmware**: examples/device/uac2_audio_headset

**Describe the bug**
On the pico the callback to fill the transfer buffer are only getting called 200-500 times a second, if I try to up the data size the callback frequency goes down.
When sending data what controls the callback frequency?

PanRe, 2021-02-12:
Every time the host collected or sent data an interrupt triggers a callback function within the audio driver in which a new transmit for receive or transmit is scheduled.

There is a new design for the audio driver on its rollout, completely redesigning how data is copied. However, i can not say when the PR is finished.

ndinsmore, 2021-02-22:
This issue was a bug caused by the pico dcd fifo implementation, and a buffer overrun.   It can now be closed.

closed 2021-02-22 by ndinsmore