hathach/tinyusb #639: MIDI Output On Other Channels Not Being Picked Up

IrisMakes, 2021-02-08:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Windows 10
 - **Board**   : Raspberry Pi Pico (RP2040)
 - **Firmware**: examples/device/midi_test

**Describe the bug**
When changing the channel used by tudi_midi_write24 in the midi_test example code, both MIDI-OX and Ableton Live don't detect anything on the channel. With channel 1, things work perfectly fine.
[image]

With any other channel however, nothing in picked up by the software. Looking at the packets with Wireshark, it looks like it is sending signals on those channels, but nothing is being picked up.
[image]

I did put this code into the pico examples folder so it built easier, but as far as I can tell that shouldn't cause a bug like this.

**To reproduce**
Steps to reproduce the behavior:
1. Change the channel for tudi_midi_write24 from 0 to 1 in both the note on and note off cases
2. Build midi_test and upload the uf2 file
3. Open MIDI-OX or Ableton Live
4. Connect to the device

**Screenshots**
[image]
where code was changed

[image]
current file structure of pico-examples project

[image]
different CMakeLists.txt file to work with pico-examples

**Log**
As far as I can tell, serial debugging doesn't work when using USB features on the PICO, and I don't have a way to wire the UART to another port.

hathach, 2021-02-09:
please git clone & build the example using this repo make, since it is what I am using for testing. It also ensures that you are building the latest code from this repo

https://github.com/hathach/tinyusb/blob/master/docs/getting_started.md#build

IrisMakes, 2021-02-09:
I cloned the repo and built the midi_test example with the changed channel, and I'm still getting the same issue. Just thought I'd give an update before I figure out how to get the log working.

IrisMakes, 2021-02-11:
Here's what I got from the log:

USBD Xfer Complete on EP 81 with 4 bytes
  MIDI xfer callback
  Queue EP 81 with 4 bytes ... OK
USBD Xfer Complete on EP 81 with 4 bytes
  MIDI xfer callback
  Queue EP 81 with 4 bytes ... OK
USBD Xfer Complete on EP 81 with 4 bytes
  MIDI xfer callback
  Queue EP 81 with 4 bytes ... OK
USBD Xfer Complete on EP 81 with 4 bytes
  MIDI xfer callback
  Queue EP 81 with 4 bytes ... OK

And then it kept repeating.

stuffmatic, 2021-02-18:
@IrisMakes As far as I can tell, the first argument to `tudi_midi_write24` is not the MIDI channel but the `jack_id`. I managed to get the example to run on different channels by setting [the low nibble](http://midi.teragonaudio.com/tech/midispec/noteon.htm) of the second argument like this

```
// Send Note On for current position at full velocity (127) on channel [CHANNEL].
tudi_midi_write24(0, 0x9[CHANNEL], note_sequence[note_pos], 127);
```

hathach, 2021-02-19:
right the first byte is cable number or jack id, the API name does have a typo somehow, I will update it maybe rename the write24() to write32() as well since 24 is a bit misleading.
[image]

hathach, 2021-02-19:
update: PR is coming https://github.com/hathach/tinyusb/pull/660 , please try to modify the `channel` variable to see if that work for you.

IrisMakes, 2021-02-21:
Looks good now! Thanks for the help!
[image]

closed 2021-02-22 by hathach