hathach/tinyusb #651: Support running entire stack safely in ISR

hathach, 2021-02-13:
**Is your feature request related to a problem? Please describe.**
For some scenario, user prefer to run entire stack in ISR in low priority mode. At current state, it may be not safe due to osal mutex/queue api that can lead to dead lock depending on the rtos, usage. 

Soluion: implement a safe **tud_task_isr()** api

HiFiPhile, 2021-02-26:
That's indeed a good idea. I've tried to run the stack in PendSV in low priority triggered by SysTick with additional mutex, in this way I don't need a RTOS abd I can have Delay() in main loop.

mfp20, 2021-07-01:
> That's indeed a good idea. I've tried to run the stack in PendSV in low priority triggered by SysTick with additional mutex, in this way I don't need a RTOS and I can have Delay() in main loop.

How fast did you call tud_task() from ISR? Every 125us to match USB2.0 microframes?

I need something similar and I was just wondering if I can call tud_task() from an ISR.

HiFiPhile, 2021-07-01:
> How fast did you call tud_task() from ISR? Every 125us to match USB2.0 microframes?

It's a STM32L0 @ 64MHz, since it's full speed so I call it at 3kHz without problem.

I haven't tried for high speed.

mfp20, 2021-07-01:
> since it's full speed so I call it at 3kHz without problem

That's more than needed; you might be good at 1kHz as USB1.1 polls each 1ms.

What about the extra mutex? Is it needed? What do you lock?

hathach, 2021-07-01:
currently it can only be safe to call tud_task() if you don't use RTOS. Otherwise calling the mutex API  within ISR can cause issues depending on your RTOS. The idea of introducing tud_task_isr() is to skip mutex API.

mfp20, 2021-07-01:
> currently it can only be safe to call tud_task() if you don't use RTOS. Otherwise calling the mutex API within ISR can cause issues depending on your RTOS. The idea of introducing tud_task_isr() is to skip mutex API.

Well, RPi Pico doesn't have an RTOS but I've spotted both mutex (irqs on) and critical_section (irqs off) used in code; don't know HOW they are used tough.

In any case yes, having a isr safe tud_task() would help a lot.