hathach/tinyusb #656: make failing on cygwin

pran005, 2021-02-17:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   :  Windows 10
 - **Board**   : 
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**
Make fails to find the source files when fired from cygwin terminal. 
CC board.o
arm-none-eabi-gcc.exe: error: /%localdir%/tinyusb/hw/bsp/board.c: No such file or directory
**To reproduce**
Steps to reproduce the behavior:
1. Open cygwin terminal (with make installed)
2. Run make -C examples/device/cdc_msc BOARD=<board_name>
3. See error

**Screenshots**

[image]

pran005, 2021-02-17:
Hi @hathach  what development environment would you suggest if I wish to contribute to this project? I am currently using a Windows 10 PC, with make installed on cygwin terminal.  Would you prefer me to build this project on a unix machine?

cr1901, 2021-02-18:
I assume that your `C:\Users\Pranjal\Documents\...` etc isn't reachable from cygwin's root (`/`, `/home`, `/bin`, `/etc`). To access `C:\Users\Pranjal\Documents\...` from cygwin, you need to use `/cygdrive/c/Users/Pranjal/`, not `/c/Users/Pranjal/`. I'm not sure the best way to put that into the Makefile.

pran005, 2021-02-18:
Thanks for the suggestion but unfortunately, the path you mentioned i.e. /cygdrive/c/Users/Pranjal/ , was causing trouble initially, so I aliased it to /c/Users/Pranjal. You're right that these are not the best paths to put into Makefile, I figured out the issue to be the $(TOP) in the top.mk. (The other makefiles that work use relative paths). Currently, I am able to build it under /home but was hoping if there could be a better solution.

Thanks

hathach, 2021-02-18:
Although I don't use windows, I guess WSL probably works better than cygwin.

pran005, 2021-02-18:
Thanks for the suggestion @hathach ! Can you tell me a preferred development environment if I wish to contribute to the stack? Shall I use Linux and ditch windows?

hathach, 2021-02-18:
you should use any environment you feel comfortable with. It is not too important at all.

pran005, 2021-02-18:
Alright. Thanks.  One more question @hathach , a little unrelated to this topic. Is it absolutely necessary to use the vendor provided drivers as a BSP if I m porting it to a new MCU?

hathach, 2021-02-18:
> Alright. Thanks. One more question @hathach , a little unrelated to this topic. Is it absolutely necessary to use the vendor provided drivers as a BSP if I m porting it to a new MCU?

You could use any board you have, I tend to go for common/popular one since that is easier for people to get and test.

kkitayam, 2021-02-18:
If you are okay with using a non-Cygwin system, examples can be built with msys2 or windows command prompt. I had confirmed at #590 

I guess that [arm gcc](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm) for windows was built with msys2. If gcc was built with cygwin, examples could be built by the gcc on cygwin.

kkitayam, 2021-02-18:
I found the same issue on [stackoverflow](https://stackoverflow.com/a/34052959).
It said that [arm gcc](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm) for windows does not support with cygwin make.

kkitayam, 2021-02-18:
I suggest msys2 or command prompt for windows users, because [WSL2 does not yet support USB devices](https://docs.microsoft.com/en-us/windows/wsl/wsl2-faq#can-i-access-the-gpu-in-wsl-2-are-there-plans-to-increase-hardware-support). WSL2 can build samples, but it cannot flash them via J-Link. Docker for windows does the same.

pran005, 2021-02-19:
Thanks for this @kkitayam . It helped! I tried building it on WSL as suggested by @hathach , it seems to build. But I haven't tried flashing it using J-Link or OpenOCD + ICDI yet. Although I'm able to get that done using cygwin, I can try accessing  the same project  from WSL & Cygwin for both the things as a ad-hoc solution for now. 

As far as command prompt is concerned, I tried installing make using choco pack manager, but the results were similar to that of cygwin. Can you please suggest some other make utility that works fine on command prompt/powershell. 

Regards

kkitayam, 2021-02-20:
> As far as command prompt is concerned, I tried installing make using choco pack manager, but the results were similar to that of cygwin.

I guess that your PATH environment variable has `C:\cygwin64\bin`  for cygwin. The path for cygwin causes failure.
make for windows treats SHELL environment variable as special. Please refer to [Choosing-the-Shell](https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html)
To avoid the issue,  would you specify `SHELL=cmd.exe` on arguments like as following.
```
C:\> C:\ProgramData\chocolatey\bin\make.exe  SHELL=cmd.exe BOARD=<board_name>
```

In my investigation is followings.
We can check by `where` that which make is priority high on PATH environment.
For example, following is in my environment. 
```
C:\>where make
C:\ProgramData\chocolatey\bin\make.exe
```

I guess @pran005 's environment shows following.
```
C:\>where make
C:\cygwin64\bin\make.exe
C:\ProgramData\chocolatey\bin\make.exe
```

I tried to build cdc_msc example in above settings, then that failed.
```
C:\work\tinyusb\examples\device\cdc_msc>C:\ProgramData\chocolatey\bin\make BOARD=frdm_kl25z
/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `for %%i in ( ../../.. ) do echo %%~fi'
../../make.mk:45: *** Invalid BOARD specified.  Stop.
```
The message indicates that sh.exe is invoked as a shell because of sh.exe exists on the PATH and not defined `SHELL` environment variable.

I tried to build with `SHELL` specified explicitly, then succeeded!
```
c:\work\tinyusb\examples\device\cdc_msc>C:\ProgramData\chocolatey\bin\make SHELL=cmd.exe BOARD=frdm_kl25z
AS startup_MKL25Z4.o
CC system_MKL25Z4.o
CC clock_config.o
CC fsl_clock.o
... snip
LINK _build/frdm_kl25z/frdm_kl25z-cdc_msc.elf
CREATE _build/frdm_kl25z/frdm_kl25z-cdc_msc.bin
CREATE _build/frdm_kl25z/frdm_kl25z-cdc_msc.hex
''
   text    data     bss     dec     hex filename
  18240    8364    5104   31708    7bdc _build/frdm_kl25z/frdm_kl25z-cdc_msc.elf
''
```

pran005, 2021-02-21:
Thank You so much @kkitayam ! You're right! Specifying the shell explicitly does solve the problem! Thanks for the help! :)

closed 2021-02-21 by hathach