hathach/tinyusb #659: Mynewt build issue

jocelynmass, 2021-02-18:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : macOS 11.0.1 
 - **Board**   : -
 - **Firmware**: -

**Describe the bug**
Building tinyusb with mynewt will trig the following warning message:

`* Warning: Multiple packages with same pkg.name=tinyusb in repo tinyusb; path1=/cionic-newt-dev/repos/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb path2=/cionic-newt-dev/repos/tinyusb`

The issue is coming from pico-sdk repo in hw/raspberrypi folder, as it also includes tinyusb folder.

hathach, 2021-02-18:
yeah, I notice the issue as well. It is getting a bit complicated since `pico-sdk` include tinyusb as submodule, and tinyusb also include pico-sdk for building its example. It cross-includes, I don't understand the newt build system to have a sound suggestion. For a walkaround, you could just remove the `repos/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb` as my newt example ci  https://github.com/hathach/mynewt-tinyusb-example/blob/master/.github/workflows/build.yml#L48

jocelynmass, 2021-02-19:
Yeah I tried to get it working with mynewt without removing `repos/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb`, but so far I failed. Anyway, if I find way to avoid this issue, I'd PRd it.
Thanks

hathach, 2021-02-19:
thanks for updating, things getting a bit complicated since projects now depends on each other, making recursively submodule clone is out of control. Actually, tinyusb doesn't need any of the submodule, those are mainly for example building and testing. If you know newt have any flags to add to repsitory.yml for that, I will be happy to make changes.

jocelynmass, 2021-02-19:
Yeah, maybe examples should sit in a different repo in order to keep the USB stack clean.

hathach, 2021-02-19:
There is an src only, read only mirror here for those who need https://github.com/hathach/tinyusb_src 

Maintaining an extra examples repo and keeping then in sync is too much work for me at the moment. Other project doesn't need to clone any of the submodules.

mlaz, 2021-02-19:
@jocelynmass when porting an mcu to mynewt, if there are external dependencies these should be excluded.

Either by deleting them or setting the corresponding mcu pkg.yml via pkg.ign_files or pkg.ign_dirs.

jocelynmass, 2021-02-19:
Thanks @mlaz, good to know.

PR #782 (2021-04-11): Remove submodule pico sdk
**Describe the PR**
remove pico-sdk from submodules list, user should install pico-sdk by themself or set environment `PICO_SDK_PATH` to correct location. close #659 

related https://github.com/raspberrypi/pico-sdk/pull/321
closed 2021-04-11 by hathach