hathach/tinyusb #668: [question] on unmount callback

chegewara, 2021-02-20:
After unplug USB cable from PC i am getting this:
```
20:34:02.400 -> device unmounted
20:34:02.433 -> device unmounted
20:34:02.466 -> device unmounted
20:34:02.466 -> device unmounted
20:34:02.499 -> device unmounted
20:34:02.499 -> device unmounted
20:34:02.533 -> device unmounted
20:34:02.566 -> device unmounted
20:34:02.566 -> device unmounted
20:34:02.599 -> device unmounted
...
20:34:06.108 -> device unmounted
20:34:06.108 -> device unmounted
20:34:06.207 -> device unmounted
20:34:06.671 -> device mounted
```
Is it normal that on `void tud_umount_cb(void)` callback is invoked so often as long as the device is not mounted again?

chegewara, 2021-02-21:
Its probably user fault. After setting flag on unmount which is returning false in `tud_msc_test_unit_ready_cb` it looks much better.

closed 2021-02-21 by chegewara

hathach, 2021-02-22:
@chegewara no it is not, it should only be invoked once, which tinyusb stack you are testing with. Please make sure you run with this repo master.  esp32s2 port currently doesn't seem to detect  disconnection probably. From what I see, your version use `Suspended` for disconnected.

somewhat similar https://github.com/hathach/tinyusb/issues/564

chegewara, 2021-02-22:
Thanks, i will try to check with this repo. The iisue i have with my library on arduino esp32, which seems to be very far from original. Also it is version before 0.8.0 (with broken return value from tinyusb_init).