hathach/tinyusb #673: Readme logo invisible in dark theme

bnainar, 2021-02-23:
[image]

hathach, 2021-02-23:
Yeah, I use dark theme as well and have no idea how or which format to make it appeared nicely on both light and dark theme.

dhalbert, 2021-02-23:
Here is a quick substitute image. The background of the original png is transparent. I made it white and added 10 pixels all around, because it was cropped very tightly.

[image]

bnainar, 2021-02-23:
@dhalbert, looks good. Make a PR

hathach, 2021-02-24:
thanks @dhalbert , I have update readme with link to your image :+1: :+1:

closed 2021-02-24 by hathach