hathach/tinyusb #677: Number of audio input channels hardcoded in AUDIO_MIC descriptor

mbruno-xmos, 2021-02-24:
https://github.com/hathach/tinyusb/blob/ac30211853e9f8948d03db871f839e93a75b1edd/src/device/usbd.h#L429

I have created a two channel USB audio microphone application. I have set CFG_TUD_AUDIO_N_CHANNELS_TX to two, but I needed to modify the referenced line to set _nchannelsphysical to CFG_TUD_AUDIO_N_CHANNELS_TX rather than to 0x01.

I am wondering if I should not have used TUD_AUDIO_MIC_DESCRIPTOR in my application, but rather created my own similar macro to define exactly the descriptors that I need. Still, I think it would make sense for the number of channels set by TUD_AUDIO_MIC_DESCRIPTOR to be configurable and not hardcoded to 1.

I also needed to define CFG_TUD_AUDIO_TX_FIFO_COUNT to CFG_TUD_AUDIO_N_CHANNELS_TX in my application. By default it is defined to CFG_TUD_AUDIO_N_CHANNELS_TX inside audio_device.c, but to 1 inside audio_device.h. This results in a mismatch between the prototype and declaration of tud_audio_n_write().

PanRe, 2021-03-10:
Sorry for the late reply! The audio driver was not fully finished at your time using but it is nearly done. There will be some changes regarding the data structure with the FIFOs, but i tried to make everything easier! Please have a look on the new driver once it is out. Not everything is fully tested as audio is a pretty broad topic, so i am happy if you can test it too!

The descriptors were written hard coded at the time being. If you find the time you are very welcome to generalize the desciptor stuff! Be aware, that the descriptors must make sense with respect to the UAC2 specifications!

PanRe, 2021-04-15:
The new driver is out so if you want you may test it! The descriptors are not changed simply because the feature unit would need to be of changing size. Nevertheless, it should be pretty easy to setup your desired descriptors using the examples given ;)

closed 2021-04-15 by PanRe