hathach/tinyusb #679: CDC device read problem

wk89463, 2021-02-24:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Ubuntu 20.04 
 - **Board**   : own
 - **Firmware**: examples/device/cdc_dual_ports

**Describe the bug**
The CDC device calls read to read the data sent by the host. The data read is not the latest, but the data of the previous frame.

hathach, 2021-02-24:
please describe it more in step process, and how you test it. It doesn't sound like issue to me, all data is put to fifo and retrieved in that order.

wk89463, 2021-02-24:
USBD init
CDC init
dcd_int
dcd_connect
dcd_int_enable
USBD Bus Reset 

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Bus Reset 

USBD Setup Received 00 05 2D 00 00 00 00 00 
  Set Address
dcd_set_address 45
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Queue EP 80 with 10 bytes
USBD Xfer Complete on EP 80 with 10 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 00 02 00 00 09 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 00 02 00 00 4B 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 11 bytes
USBD Xfer Complete on EP 80 with 11 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 02 03 09 04 FF 00 
  Get Descriptor String[2]
  Queue EP 80 with 30 bytes
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 01 03 09 04 FF 00 
  Get Descriptor String[1]
  Queue EP 80 with 16 bytes
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 03 03 09 04 FF 00 
  Get Descriptor String[3]
  Queue EP 80 with 14 bytes
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
  Open EP 81 with Size = 8
dcd_edpt_open epnum 1 -- dir 1
  Open EP 02 with Size = 64
dcd_edpt_open epnum 2 -- dir 0
  Open EP 83 with Size = 64
dcd_edpt_open epnum 3 -- dir 1
  Queue EP 02 with 64 bytes ... OK
  CDC opened
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 04 03 09 04 02 00 
  Get Descriptor String[4]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 80 06 04 03 09 04 18 00 
  Get Descriptor String[4]
  Queue EP 80 with 24 bytes
USBD Xfer Complete on EP 80 with 24 bytes
  Queue EP 00 with zlp Status

USBD Setup Received 21 20 00 00 00 00 07 00 
  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes
USBD Xfer Complete on EP 00 with 7 bytes
  CDC control complete
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

/*******************************************************/
this is setup process，it's ok，when host send a data to device，the device enter interrupt and then call dcd_event_xfer_complete(0, ep_addr, xfer_size, XFER_RESULT_SUCCESS, true) ； tud_task will get it and call cdcd_xfer_cb(); the cdcd_xfer_cb first copy cdcd_interface_t->epout_buf to cdcd_interface_t->rx_ff, but now the cdcd_interface_t->epout_buf don't have data, you use _prep_out_transaction() to get data;

hathach, 2021-02-24:
I don’t understand what is your point. Please be concise why you think cdc misbehave. Also what is your mcu ?

wk89463, 2021-02-25:
now cdc process is :  device get data (dir: out) -> call cdcd_xfer_cb() -> in call cdcd_xfer_cb(), first copy data from epout_buf to rx_ff, but now in epout_buf there not have latest data, only after use _prep_out_transaction() epout_buf have the latest data. this make a problem, when I use read() to read FIFO, I always get the data of the previous frame instead of the latest one.  this is my question, is there something wrong with my understanding of the cdc out data process？my mcu is our company，it's a new mcu。

wk89463, 2021-02-25:
bool cdcd_xfer_cb(uint8_t rhport, uint8_t ep_addr, xfer_result_t result, uint32_t xferred_bytes)
{
  (void) result;

  uint8_t itf;
  cdcd_interface_t* p_cdc;

  // Received new data
  if ( ep_addr == p_cdc->ep_out )
  {
    // TODO search for wanted char first for better performance
    for(uint32_t i=0; i<xferred_bytes; i++)
    {
      tu_fifo_write(&p_cdc->rx_ff, &p_cdc->epout_buf[i]);    //  write data from epout_buf to rx_ff
    }

    // invoke receive callback (if there is still data)
    if (tud_cdc_rx_cb && tu_fifo_count(&p_cdc->rx_ff) ) tud_cdc_rx_cb(itf);

    // prepare for OUT transaction
    _prep_out_transaction(p_cdc);                                    //  write data from mcu's fifo to epout_ff
  }
}

wk89463, 2021-02-25:
Sorry, I found out. It's my problem

closed 2021-02-25 by hathach