hathach/tinyusb #681: Better way to handle CDC_REQUEST_SEND_BREAK

amitv87, 2021-02-25:
Found few legacy apps on windows which try to send break requests and terminate the connection if request fails.

Meanwhile I have made a change in **cdc_device.c** to support **CDC_REQUEST_SEND_BREAK** by adding a new case as show in code snippet.
``` c
case CDC_REQUEST_SEND_BREAK:
  if (stage == CONTROL_STAGE_SETUP)
  {
    tud_control_status(rhport, request);
  }
  else if (stage == CONTROL_STAGE_ACK)
  {
    TU_LOG2("  Send Break\r\n");
    if ( tud_cdc_send_break_cb ) tud_cdc_send_break_cb(itf);
  }
break;
```

hathach, 2021-02-25:
Can you name those legacy apps, and how to reproduce the request.

amitv87, 2021-02-25:
I have setup an rpi pico to behave as USB to UART bridge using CDC. This works perfectly fine in almost all scenarios.

I encountered the behaviour when I tried to flash a firmware using [qflash](https://forums.quectel.com/uploads/short-url/ApQc64SUOmNIU7ibGcQNPB5prxD.pdf) on a quectel modem.
The tool works absolutely fine with ftdi, silabs, prolific etc.

Although the modem needs just the rx and tx uart lines, qflash blindly expects the break request to be acknowledged else it chooses close the connection and halts the progress. It sends break request right after the connection is opened.

Also I have no idea about any standard use case of break request.

hathach, 2021-02-25:
thanks for the context, I am ok with adding more callback, but I would also like to take the chance to understand what the request does as well. I am currently working on other works, will pull out the cdc specs later on. Meanwhile, you could submit an PR for modification (it looks good), since I prefer to review/discuss code modification via PR than issue.

amitv87, 2021-02-25:
Great, will send a pr asap.
Also thanks for writing this wonderful library.
👍

hathach, 2021-02-26:
no problem, I am glad you find this project useful. closed by #683

closed 2021-02-26 by hathach