hathach/tinyusb #694: get device descriptor error

geek-slt, 2021-03-03:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   :Windows 10
 - **Board**   : STM32 F407VGXX
 - **Firmware**: examples/device/hid_composite_freeRTOS

**Describe the bug**
0.Projects created using the master branch on 2.24 can run normally
1.tusb_init() success  
2.The windows device manager prompts that the device descriptor request failed
3.It looks like uint8_t const * tud_descriptor_device_cb(void) function is not executed

**Screenshots**
If applicable, add screenshots, bus capture to help explain your problem. 
[image]

geek-slt, 2021-03-04:
I tested it in depth and found that the problem appeared in the loop of tud_task(), so I added log output to the tud_task() function, and found that the program can enter the tud_task()->while(1) loop and be able to get from the queue I received the event, but it is strange that the DCD_EVENT_BUS_RESET event can be received, but the DCD_EVENT_SETUP_RECEIVED cannot be received. This is why the process_control_request() cannot be executed and cannot respond to the host's descriptor request, which causes windows to prompt me that the device descriptor request failed. Knowing the above but not sure about the next operation, can you please help me?

geek-slt, 2021-03-04:
[image]

geek-slt, 2021-03-04:
[image]

hathach, 2021-03-04:
I am not familiar with your IDE and set up. Look like you are having some integration issue which is not really qualified as stack bug. Which I have no idea how to fix. I will move it to discussion, maybe others with similar ide set up could help

closed 2021-03-04 by hathach