hathach/tinyusb #709: RP2040 Host Stack fails with HID devices

ristohelkio, 2021-03-09:
**Set up**
 - **Board**   : Raspberry Pi Pico with UART tracing via Raspberry Pi 4
 - **Firmware**: /examples/host/cdc_msc_hid
 - **Keyboard**: Generic wireless keypad (Exibel - Provision Technology, Inc)
 - **HUB-keyboard**: Apple

**Describe the bug**
When connecting a keyboard to Pico the process fails at the point where the stack is trying to read the full Device Descriptor. If the Device Descriptor fits into one single packet (bMaxPacketSize0 == 64) the reading is successful. When bMaxPacketSize0 == 8 the stack fails.

**To reproduce**
Connect a keyboard to pico

**Log**
Exibel keypad. When the stack is trying to read an 18 bytes long Device Descriptor it fails. Result is PANIC with 'ep 0 in was already available'
[trace-exibel-keypad.txt](https://github.com/hathach/tinyusb/files/6109326/trace-exibel-keypad.txt)

Additional logs added in file src/portable/raspberrypi/rp2040/rp2040_usb.c (pico_trace replaced with TU_LOG2). The stack seems to receive 8 + 8 + 2 bytes but results in PANIC with 'Data Seq Error ':
[trace-exibel-keypad-detailed.txt](https://github.com/hathach/tinyusb/files/6109385/trace-exibel-keypad-detailed.txt)

In comparison a trace with an Apple keyboard-hub. The hub is connecting properly but when trying to read the Device Descriptor of the keyboard behind the hub, the stack fails. Hub bMaxPacketSize0 is 64 which seems to do the trick.
[trace-apple-keyboard-hub.txt](https://github.com/hathach/tinyusb/files/6109426/trace-apple-keyboard-hub.txt)

hathach, 2021-03-09:
thanks for the issue, make sure you are running the example with latest master. Though I am not surprized that it doesn't work. It is not your setup or device, it is just the host stack isn't received much of developing time. So far I have reworked usbh and host msc, will do host hid next when I find time.

hathach, 2021-05-23:
probably be fixed by  raspberrypi/tinyusb#7 based approach.

PR #862 (2021-06-01): Fix host rp2040 with low speed device
**Describe the PR**
Fix host rp2040 when mounting low speed device (e.g keyboard) with max packet size = 8
- Correct Data Toggle by number of packet: e.g  64 bytes = 8 x 8 bytes --> Data Toggle should not be flipped in the next buffer.  
- Improve RP2040-E4 (host buf sel) fix to prevent panic with already available.

Fix #709, also replace https://github.com/raspberrypi/tinyusb/pull/7 as the fix
closed 2021-06-01 by hathach