hathach/tinyusb #718: tud_network_mac_address should be initialized by hardware ID but is const

maxnet, 2021-03-14:
**Is your feature request related to a problem? Please describe.**

https://github.com/hathach/tinyusb/blob/master/examples/device/net_lwip_webserver/src/main.c#L61 mentions that ideally the MAC should be generated by hardware ID.
However the variable is declared const in https://github.com/hathach/tinyusb/blob/master/src/class/net/net_device.h#L62 and in  https://github.com/hathach/tinyusb/blob/master/lib/networking/rndis_reports.c#L39

**Describe the solution you'd like**

Remove "const" so that the MAC can be changed by program in an initialization routine at the start of program that queries the device's serial number and derives MAC from that.

hathach, 2021-03-17:
yeah, it even has an TODO remove later as well, apparently I plan to remove it. I will check it out later on to see what is best to change.

MattMills, 2023-03-22:
@hathach Any progress on this? Having an address that needs to be different on every board be const seems like a bad plan...

If there is currently no plans to move the TODO to TODONE, is there any chance the const can be removed?

hathach, 2023-03-22:
I haven't got time to reviewt net driver, please submit PR if you have suggestion.

hathach, 2023-03-29:
fixed now

closed 2023-03-29 by hathach

Slion, 2024-09-07:
I'm pretty sure this should be reopened, see: https://github.com/hathach/tinyusb/pull/1971#issuecomment-2335141214

HiFiPhile, 2024-09-07:
> Proper fix should be to use a callback.

If you are up to something a PR is welcomed.

Slion, 2024-09-07:
> If you are up to something a PR is welcomed.

Currently working with an older version of TinyUSB but I'll try to come up with a PR at some point.

Slion, 2024-09-07:
Ok the fix works if you init the variable early enough. I updated my comment there: https://github.com/hathach/tinyusb/pull/1971#issuecomment-2335141214