hathach/tinyusb #735: midi: bad things happen if device doesn't read midi from host

jepler, 2021-03-18:
We were testing midi on feather m7 rt1011, and @ladyada noticed that if MIDI was being sent out but not acted on by CIrcuitPython, "bad things happened".  On Linux this manifested as being unable to ctrl-c in the REPL and getting messages in the kernel log: `sound midiC4D0: rawmidi drain error (avail = 3934, buffer_size = 4096)`.  Would probably need to throw out some data if buffers fill up?  discarding is better than crashing

hathach, 2021-03-19:
yeah, we kind of needing a way to drain the midi buffer. However as notice by MIDI user in #618 , dropping data will cause issue with sysEx message rendering the driver unreliable. In practice,  it is rare to have MIDI user that only receive data and do nothing with the data, possibly only happens with test example.

ladyada, 2021-03-19:
thats ok but it should not take down entire USB subsystem :D 
perhaps it can release unlistened packets?

hathach, 2021-03-19:
@ladyada hmm, taking down entire stack is not nice, maybe it got some assert() and hanged. I will check it out.

ladyada, 2021-03-19:
yeah our imx board stopped responding altogether

PR #766 (2021-04-02): Enhance midi
**Describe the PR**
Fix #735. should also fix root cause of #377 . should also fix issue with https://github.com/adafruit/circuitpython/issues/4190
Root cause of the issue is TX fifo is overflow, though it is not handled safely by driver, which cause the data corruption to the usbd device control and cause the system to crash/hardfault. While fixing the issue, I have go through the whole improvement, including
- Adding MIDI_CIN_ and MIDI_STATUS_  instead of using magic number in stream API
- Rename read(), write() to stream_read(), stream_write() with deprecated warning
- Rename r [... cut]
closed 2021-04-02 by hathach