hathach/tinyusb #746: cdc_device DTR bit check breaks popular mcu flash tools, should be optional in src/class/cdc/cdc_device.c

eriklundh, 2021-03-23:
**Set Up**

- **Raspberry pico-sdk 1.1 with 
[cdc_device.zip](https://github.com/hathach/tinyusb/files/6186099/cdc_device.zip)

tinyusb @ 11c23f8** 
- **Raspberry Pico** 
-  **src/class/cdc/cdc_device.c** 

**Describe The Bug**

cdc_device.c throttles serial data if DTR is false
ESP32, flashing tools such as esptool.py assume no flow control and use DTR and RTS to reset ESP32 and put it in flash mode. 
When e g Raspberry Pico is used as a USB to serial converter to ESP32, the current cdc_device.c behavior interfere with esptool.py, which expect serial data to flow back and forth while it resets ESP32 by changing the state of DTR and RTS:

Current function in src/class/cdc/cdc_device.c in Raspberry pico-sdk 1.1, which causes the problem
bool tud_cdc_n_connected(uint8_t itf)
{
  // DTR (bit 0) active  is considered as connected
  return tud_ready() && (tu_bit_test(_cdcd_itf[itf].line_state, 0) );
}

// This minimal change solves the problem, but merely disables the DTR logic and is not configurable
bool tud_cdc_n_connected(uint8_t itf)
{
  // DTR (bit 0) active  is considered as connected
  return tud_ready() && (tu_bit_test(_cdcd_itf[itf].line_state, 0) **|| true**);
}

I would be happy to fork, introduce my changes, and send a pull request.
But I would be grateful for guidance if a compile-time flag is preferred by the tinyusb maintainers, or should I introduce a flag variable that controls if DTR is checked at runtime? (flag as another field in cdcd_interface_t _cdcd_itf). A flag would need get(itf) and set(itf,state) functions since cdcd_interface_t is instantiated as a (private) static _cdcd_itf in cdc_device.c

duempel, 2021-03-23:
Thanks for your information. There was actually a talk about ESP32 flashing before starting [here](https://github.com/hathach/tinyusb/pull/401#issuecomment-656717740).

CDC already changed that it's possible to receive and transmit data even if DTR bit is not set. The `tud_cdc_n_connected` API still checks this bit because some application might require a real terminal connection to be detected. That's why it should be kept within this function.

I guess in your case you just do not have to call `tud_cdc_connected`.  The previous examples still made use of this (check [this](https://github.com/hathach/tinyusb/pull/506#discussion_r486533305)) but I just noticed @hathach changed the cdc example sources 5 days ago. If you now take a look to device example's `cdc_task` it's not checked anymore. 😉

hathach, 2021-03-23:
it is already optional, have you tried the code at master instead of the one come with pico-sdk ?