hathach/tinyusb #750: EFM32 Driver: error -110 upon first enumeration

perigoso, 2021-03-23:
> Ok I have now tested a couple different examples, all without issues _after_ enumeration, but there is one common issue, every time after initialization the devices times out
> 
> ```
> [ 5845.144641] usb 3-4.3: USB disconnect, device number 25
> [ 5889.104950] usb 3-4.3: new full-speed USB device number 26 using xhci_hcd
> 
> [ 5904.562516] usb 3-4.3: device descriptor read/64, error -110
> 
> [ 5904.767196] usb 3-4.3: New USB device found, idVendor=cafe, idProduct=4004, bcdDevice= 1.00
> [ 5904.767203] usb 3-4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 5904.767206] usb 3-4.3: Product: TinyUSB Device
> [ 5904.767210] usb 3-4.3: Manufacturer: TinyUSB
> [ 5904.767213] usb 3-4.3: SerialNumber: 123456
> ```
> this happens every time after reset, but after the first enumeration, the device re-enumerates fine, as in, unplugging and plugging back in, it's immediate (device is self powered for testing)
> 
> not quite sure how to debug this, I dont have a capable enough logic analyzer, I've looked at wireshark but no packets show up before the timeout, everything afterwards looks normal.

From > #703

perigoso, 2021-05-06:
@hathach do you have he `sltb009a` yet? I now have a [luna](https://github.com/greatscottgadgets/luna) so I can capture some lower level packets, I want to tackle this bug, because its a bit annoying having to wait 10s every time i plug in a device :stuck_out_tongue:

hathach, 2021-05-07:
I still haven't got the board yet, I have some back order from digikey. and They just shipped it out. Though I currently don't have much time to work on this atm, but I will try to help you if needed.

hathach, 2021-11-04:
I have tested with sltb009a and didn't notice this issue, please give it a try again with the new dwc2 driver.

perigoso, 2021-11-04:
Thanks for testing, I will try again later today

perigoso, 2021-11-04:
Thanks for testing, I will try again later today.

and good job with the generalization of the driver

perigoso, 2021-11-04:
The issue is in fact fixed! good job.

I didn't investigate much, so I don't know what was causing it, just confirmed it was still present, and it's fixed as of the `dwc2` merge

my system: `Linux 5.14.12-zen1-1-zen x86_64` just for reference, I never tested this on windows or mac

closed 2021-11-04 by perigoso

hathach, 2021-11-05:
thank you for testing it out