hathach/tinyusb #755: Buildsystem does not allow a c and assembly source with the same name

perigoso, 2021-03-27:
Buildsystem does not allow a c and assembly source with the same name, where only one is supposed to be compiled. 

I found this issue while trying to implement an example for the efm32 family in #749. 

The Cmsis provided by silabs as both a c implementation and an assembly implementation for the startup, only one should be compiled, but the buildsystem right now tries to compile both, regardless of which is included. 

I have followed up the issue and think it is in `rules.MK` where both c and s files are compiled into a single object list, then when the objects are called for compilation, it resolves to both a c file and a s file. 

I will be working on a fix soon.

hathach, 2021-03-29:
this is indeed tricky, maybe we could auto add suffix _asm or _s into the object file of `.S` file in the rule. 
https://github.com/hathach/tinyusb/blob/master/examples/rules.mk#L55 . I will also try to make an PR to fix this as well

perigoso, 2021-03-29:
Hm i was going to go with a different approach, essentially making two different object lists, i think that would be prefered instead of renaming the objects, i will do a PR in a bit, you can then choose what you prefer, i think your approach would fix either way

perigoso, 2021-03-30:
You PR looks goog! I just tested it and got a successful build.

PR #760 (2021-03-30): add suffix _s to object of assembly file
**Describe the PR**
Fix #755 where there is .c and .s with the same name in the same folder.
closed 2021-03-30 by hathach