hathach/tinyusb #761: CDC: tud_cdc_write returns 0

ned-pcs, 2021-03-29:
**Set Up**
- Ubuntu 20.04
- STM32F072-EVAL board
- Firmware: custom app with just CDC, sending 22-byte messages as fast as possible. Using RTT for logging.
  - app waits until host connection before sending anything.

**Describe The Bug**
After 154 characters are sent successfully (7x22 byte messages), `tud_cdc_write()` returns 0 bytes written when 2 were requested.
The immediately prior `tud_cdc_write()` of 22 bytes returned 20 bytes written.

When `tud_cdc_write()` returns 0 bytes:
```
_cdcd_itf[0].tx_ff is full
line_state = 3
_usbd_dev.connected,addressed = 1
_usbd_dev.suspended = 0
ep_status[2][0] and ep_status[2][1] busy=true, stalled=false, claimed=true
_usbd_dev.cfg_num = 1 (mounted)
```

**To Reproduce**
- start application
- connect simple serial terminal program to `/dev/ttyACMx`

**Log**
```
USBD init
CDC init
USBD Bus Reset 
USBD Bus Reset 

USBD Setup Received 00 05 27 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 08 00 
  Get Descriptor Device
  Queue EP 80 with 8 bytes ... OK
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ... OK
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 02 00 
  Get Descriptor String[2]
  Queue EP 80 with 2 bytes ... OK
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 16 00 
  Get Descriptor String[2]
  Queue EP 80 with 22 bytes ... OK
USBD Xfer Complete on EP 80 with 22 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 02 00 
  Get Descriptor String[1]
  Queue EP 80 with 2 bytes ... OK
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 0E 00 
  Get Descriptor String[1]
  Queue EP 80 with 14 bytes ... OK
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 02 00 
  Get Descriptor String[3]
  Queue EP 80 with 2 bytes ... OK
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 12 00 
  Get Descriptor String[3]
  Queue EP 80 with 18 bytes ... OK
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 09 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes ... OK
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 4B 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ... OK
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 11 bytes ... OK
USBD Xfer Complete on EP 80 with 11 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
  Open EP 81 with Size = 8
  Open EP 02 with Size = 64
  Open EP 82 with Size = 64
  Queue EP 02 with 64 bytes ... OK
  CDC opened
  Queue EP 80 with 0 bytes ... OK
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 04 03 09 04 02 00 
  Get Descriptor String[4]
  Queue EP 80 with 2 bytes ... OK
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 04 03 09 04 1C 00 
  Get Descriptor String[4]
  Queue EP 80 with 28 bytes ... OK
USBD Xfer Complete on EP 80 with 28 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 03 01 00 00 00 00 00 
  Set Feature
  Queue EP 80 with 0 bytes ... OK
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 08 00 00 00 00 01 00 
  Get Configuration
  Queue EP 80 with 1 bytes ... OK
USBD Xfer Complete on EP 80 with 1 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes ... OK
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with 0 bytes ... OK
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 21 20 00 00 00 00 07 00 
  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes ... OK
USBD Xfer Complete on EP 00 with 7 bytes
  CDC control complete
  Queue EP 80 with 0 bytes ... OK
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 21 22 03 00 00 00 00 00 
  CDC control request
  Queue EP 80 with 0 bytes ... OK
USBD Xfer Complete on EP 80 with 0 bytes
  Set Control Line State: DTR = 1, RTS = 1
  Queue EP 82 with 64 bytes ... OK 		// 3 messages
USBD Xfer Complete on EP 82 with 64 bytes
  CDC xfer callback
  Queue EP 82 with 24 bytes ... OK		// 1 message
USBD Xfer Complete on EP 82 with 24 bytes 
  CDC xfer callback
  Queue EP 82 with 22 bytes ... OK		// 1 message
USBD Xfer Complete on EP 82 with 22 bytes 
  CDC xfer callback
  Queue EP 82 with 44 bytes ... OK		// 2 messages

--- and then tud_cdc_write() returns 0 for an attempted 2 byte write (after an attempted 22 byte write returned 20 bytes written).

Linux host app rec'd 7x 22-byte messages total from /dev/ttyACM0.
So last 44 bytes enqueued on EP 82 never got an xfer complete callback.
```

hathach, 2021-03-30:
there is no way for others to reproduce this issue since it is custom app. Please try to have minimal modification to the existing stock example and post your changes here.

ned-pcs, 2021-03-30:
I agree, this is not a bug. I stated it badly (clearly one can't transmit at infinite speed).
Sorry to bother you.
My problem is that the overall CDC transmission throughput (in my app that acts as a CAN-to-slcan gateway) is not high enough in practice to keep up with messages being sent rapidly on a 250kbps CAN bus, though I had hoped that it would be.
I've added code to check FIFO space using `tud_cdc_write_available()` and drop packets if not enough space is available.
Perhaps double-buffering would help, though I am not familiar enough with either TinyUSB or the STM32 USB peripheral presently (and don't have enough time) to offer a pull request to add double buffering.
Thanks for TinyUSB; it's been quite easy to use and will be easy to maintain in my app.

closed 2021-03-30 by ned-pcs

hathach, 2021-03-30:
try to increase both `CFG_TUD_CDC_EP_BUFSIZE` and `CFG_TUD_CDC_RX_BUFSIZE`, `CFG_TUD_CDC_TX_BUFSIZE` to see if it is improved.

ned-pcs, 2021-03-30:
Thanks for your attention. I don't think it'll be necessary to do that.
It turns out that what was happening was that there was not quite enough space for a whole message.
I added a check of `tud_cdc_write_available()`, and if that said that there wasn't enough space for a whole message, I did a `tud_cdc_write_flush()`.
That fixed the issue!

hathach, 2021-04-02:
Thanks for the update, I am sure this will help others