hathach/tinyusb #791: Use different build directories for debug and release builds

kapacuk, 2021-04-18:
Currently, when you build an example, the results are stored under `_build/$(BOARD)/`, regardless of the value of the `DEBUG` variable. So, if you have a fully compiled release build and type `make DEBUG=1` nothing happens. 

I think the debug and release builds should be kept in separate directories, and propose this change to the makefile:

```
--- a/examples/make.mk
+++ b/examples/make.mk
@@ -3,8 +3,11 @@
 # ---------------------------------------
 
 # Build directory
-BUILD := _build/$(BOARD)
-
+ifeq ($(DEBUG), 1)
+BUILD := _build/$(BOARD)/debug
+else
+BUILD := _build/$(BOARD)/release
+endif
 PROJECT := $(BOARD)-$(notdir $(CURDIR))
 BIN := $(TOP)/_bin/$(BOARD)/$(notdir $(CURDIR))

```
If you agree I can submit a PR.

hathach, 2021-04-19:
It is nice suggestion, though to be honest, I don't feel the need to implement it. Do you have any particular issue with current build. Normally just do a clean would work. `DEBUG=1` currently only change the optimization level. So no, I don't we should implement this.

kapacuk, 2021-04-19:
I agree, it's not a big deal, just a bit unusual - many projects do keep debug and release builds separately, so for somebody who just start using TinyUSB it might look counter-intuitive. Anyway, I'm not going to insist, closing the issue now.

closed 2021-04-19 by kapacuk