hathach/tinyusb #799: No callback for Application to handle DCD_EVENT_BUS_RESET

CraigHutchinson, 2021-04-21:
During an EMC test the USB bus event DCD_EVENT_BUS_RESET is triggering. However the application code is using the callbacks to determine the line state of as CDC device. 

As no callback event occurs the application cannot detect this event. The application must explicitly poll `tud_cdc_connected()` as a workaround which is not ideal.

https://github.com/hathach/tinyusb/blob/c5f440a6aa39a09487000e7ce408d252dc7641f0/src/device/usbd.c#L466

I am not sure if there needs to be a new callback or if the 'reset()' handler needs to cause existing handler to be triggered:
- tud_umount_cb
- tud_suspend_cb

hathach, 2021-04-21:
reopen issue when you filled out all the detail in template

CraigHutchinson, 2021-04-21:
@hathach Creating 'New issue' from the line of code does not produce a template it would appear... I have got it now via "new issue" button and will apply. cheers.

CraigHutchinson, 2021-04-21:
@hathach I have updated the description, I do not have re-open permissions though

CraigHutchinson, 2021-04-21:
on further investigation the cause of the reset may not be normal as it is possible the 'host' in this case is not aware the Reset had occured. In this situation the Device application needs to disconnect/reconnect to properly re-establish the USB connection. I will continue learning and acquiring understanding of the USB stack. 
[image]