hathach/tinyusb #802: tud_hid_descriptor_report_cb() does not provide the length of the report descriptor

dhalbert, 2021-04-23:
The HID report descriptor is supplied dynamically by `tud_hid_descriptor_report_cb()`, but the length is not passed back. The report descriptor does not include its own length. The length has to have been set previously in `p_hid->report_desc_len`. It is set in `hidd_open()`, since the length must be set in the HID descriptor (not the HID report descriptor).

I am not sure what the utility of this callback is, then. You could substitute an HID Report Descriptor of identical length, but that's all. Maybe I am missing something.

hathach, 2021-04-24:
The callback does not include the length on purpose. The length is already decided by hid descriptor (after hid interface) as part of configuration descriptor. I.e by the time `tud_descriptor_configuration_cb()` is called, you shouldn't change your report descriptor (or it has to be the same length)

[image]

hathach, 2021-04-24:
I have done some dynamic descriptors for arduino as well, you may want to take a look at
- Core https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/master/Adafruit_USBD_Device.cpp#L116
- lib https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/Adafruit_USBD_HID.cpp#L73

hathach, 2021-05-04:
@dhalbert  Let me know if you still have question, or API does not make sense to you.

closed 2021-05-12 by hathach