hathach/tinyusb #812: Please provide all details at least for Setup/Describe/Reproduce

Makurisan, 2021-04-30:
I tried your stack with samg55 in host mode with keyboard and a 4 port hub. 
The usb hub is enumerated not completely correct  and an attached keyboard devices stops at GetDeviceDescriptor. 

[image]

Makurisan, 2021-04-30:
I dont know why but the GetPortStatus is not completed:

00> HUB Get Port Status: addr = 1 port = 5
00> Control Setup: A3 00 00 00 05 00 04 00 

Last ouput in the log.

hathach, 2021-05-01:
Please fill out all the detail in bug template in the first post. It is mandatory. This issue will then be reopened

Makurisan, 2021-05-01:
**Set Up**

- **PC OS**
 Windows 10
- *Board** 
  Samg55
- **Firmware** 
 4 Port Hub with keyboard attached

**Describe The Bug**
If I attach the Hub and the keyboard the enumeration stops at GetDeviceDescriptor
**To Reproduce**

	tusb_init(); // initialize tinyusb stack

	while (true) {
		tuh_task();
	}
Steps to reproduce the behavior:
1.  plug in the hub
2.  plug in the keyboard
3. See error  stop on GetPortStatus

**Screenshots**
[image]

If applicable, add screenshots, bus capture to help explain your problem. 

**Log**

00> Get 9 bytes of Configuration Descriptor
00> Control Setup: 80 06 00 02 00 00 09 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  09 02 19 00 01 01 00 E0 32                       |........2|
00> on EP 00 with 4294967295 bytes
00> Control Setup: 80 06 00 02 00 00 19 
(Connection lost)
00> SEGGER J-Link V6.99a (beta) - Real time terminal output
00> J-Link EDU Mini V1 compiled Feb 18 2021 11:25:23 V1.0, SN=801011341
00> Process: atbackend.exe
00> HID init
00> HUB init
00> USBH DEVICE ATTACH
00> Get 8 byte of Device Descriptor
00> Control Setup: 80 06 00 01 00 00 08 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  12 01 00 02 09 00 00 08                          |........|
00> on EP 00 with 4294967295 bytes
00> Port reset 
00> Set Address 
00> Control Setup: 00 05 01 00 00 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> Control Setup: 80 06 00 01 00 00 12 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  12 01 00 02 09 00 00 08 E3 05 06 06 02 07 01 02  |................|
00>   0010:  00 01                                            |..|
00> on EP 00 with 4294967295 bytes
00> Get 9 bytes of Configuration Descriptor
00> Control Setup: 80 06 00 02 00 00 09 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  09 02 19 00 01 01 00 E0 32                       |........2|
00> on EP 00 with 4294967295 bytes
00> Control Setup: 80 06 00 02 00 00 19 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  09 02 19 00 01 01 00 E0 32 09 04 00 00 01 09 00  |........2.......|
00>   0010:  00 00 07 05 81 03 01 00 FF                       |.........|
00> on EP 00 with 4294967295 bytes
00> HUB open
00> Set Configuration Descriptor
00> Control Setup: 00 09 01 00 00 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> Device configured
00> HUB set config itf = 0
00> Control Setup: A0 06 00 00 00 00 09 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959228 bytes
00> Control data:
00>   0000:  09 29 04 09 00 16 64 00 FF                       |.)....d..|
00> on EP 00 with 4294967295 bytes
00> Control Setup: 23 03 08 00 01 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> Control Setup: 23 03 08 00 02 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> Control Setup: 23 03 08 00 03 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> Control Setup: 23 03 08 00 04 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> on EP 81 with 4294959213 bytes
00> HUB xfer callback
00> Port Status Change = 0x08
00> HUB Get Port Status: addr = 1 port = 3
00> Control Setup: A3 00 00 00 03 00 04 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959212 bytes
00> Control data:
00>   0000:  01 03 01 00                                      |....|
00> on EP 00 with 4294967295 bytes
00> HUB Clear Port Feature: addr = 1 port = 3, feature = 16
00> Control Setup: 23 01 10 00 03 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> HUB Reset Port: addr = 1 port = 3
00> Control Setup: 23 03 04 00 03 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> USBH DEVICE ATTACH
00> HUB Get Port Status: addr = 1 port = 3
00> Control Setup: A3 00 00 00 03 00 04 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  03 03 10 00                                      |....|
00> on EP 00 with 4294967295 bytes
00> HUB Clear Port Feature: addr = 1 port = 3, feature = 20
00> Control Setup: 23 01 14 00 03 00 00 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294967295 bytes
00> Get 8 byte of Device Descriptor
00> Control Setup: 80 06 00 01 00 00 08 00 
00> on EP 00 with 4294963012 bytes
00> on EP 00 with 4294959176 bytes
00> Control data:
00>   0000:  12 01 00 02 00 00 00 08                          |........|
00> on EP 00 with 4294967295 bytes
00> Port reset 
00> HUB Reset Port: addr = 1 port = 3
00> Control Setup: 23 03 04 00 03 00 00 00 
00> on EP 00 with 4294963012 bytes
00> HUB Get Port Status: addr = 1 port = 3
00> Control Setup: A3 00 00 00 03 00 04 00

hathach, 2021-05-04:
Thanks for updating the issue following the template. The SamG55 host driver is unfortunately not supported yet. I am surprised that you could get this far. Are you implementing the host driver locally. Looking at the log, look like the stack could recognize the hub and usb device attach event and able to communicate up to getting 8 bytes of device descriptor. The code that run up to the log is at https://github.com/hathach/tinyusb/blob/master/src/host/usbh.c#L745.

1. Did you implement the `hcd_frame_number()` to return the frame number ? it is currently used to perform reset delay (not ideally, but it is WIP).  
2. Also the screenshot is not really helpful, it only stop at the hub enumeration, and not the device enumeration afterwards where there is 8 byte of device descriptor is requested. Could you re-upload the screenshot including that part.   
3. Have you tried to test with another hub just to check. Host stack is still WIP.
4  The transfer bytes is way off `4294967295`, but it looks like a minor mis-reprenstation of integer size. Please fix that to have a cleaner log

Makurisan, 2021-05-04:
1. No I didnt implement or changed on hcd_frame_number.
2.  Yes, I do it if I saved my tests.
3.  Yes, I did test it with different hubs but always the same result.

I come nearer to the problem. In the file usbh.c at line 739 you initiate the reset after the 8 byte descriptor and normally the enumeration comes back with SetAddress after the second reset. This does not happen.  What happens is that the stack blocks.  I see on the code that you have had problem with the Reset and therefore you make this tuh_task inside the completion.  This is in my opinion not good and should be cleaned. I see on your comment that you had this also in mind.
I have implemented on the samd55 a complete host with multi hub support and there it works fine with port reset and port get status. I don't know what is here the problem. Do you have any idea?

Makurisan, 2021-05-04:
[image]

As I mentioned after the call of port Reset from usbh.c at line 739 and following lines.

hathach, 2021-05-04:
@Makurisan try to implement the hcd_uframe_number(), it is probably the issue, the reset must stay long enough. Your problem wihtout implement it is 
1. osal_task_delay() (based on frame number): is not last long enough
2. tuh_task() is called, but the port change status is not occurred yet. This may mess up with the follow up call.

hcd_uframe_number() is easy to implement, just return the current micro-frame number. Since SAMG is full-host, you need to multiple norma frame number it with 8. the `tuh_task()` call is a bit hacky, I am fully aware, just doesn't have the time to fix it. Will do  it next time when I work with host.

Makurisan, 2021-05-05:
Hm, I think it is long enough. In the usb 1.1 spec it is said that a port reset on a hub is from 10ms to max 20ms.
If I do the following

[image]

and

[image]

then the stack enumerates completely.

Makurisan, 2021-05-05:
Which board you use normally to check the OHCI code?

hathach, 2021-05-05:
@Makurisan oh, I just realize, samg also use OHCI. I normally test with LPC1769 for OHCI, but it has been quite some time. Currently I mostly use lpc18xx EHCI when working on host stack. I really want to improve the host stack, but couldn't find the time to so. I will for sure pull out the samg to test with OHCI next time I work on it.

Makurisan, 2021-05-05:
Yes, it is really interesting that the samg55 use OHCI. I tested it because I want low speed. I don't know if you know but low speed has some specialties and not all MCUs are fully compatible nowadays.

Makurisan, 2021-05-05:
Now I came up width the following solution:

[image]

[image]

hathach, 2021-05-06:
I am glad you figure a way to get it work. Since I will revise the code anyway later on, I pay more attention to this case.