hathach/tinyusb #842: Usb host: controller pad and midi host

jhusak, 2021-05-20:
I kindly ask to add usb host support for game controllers and midi. There is a big niche in retrogaming and retrocomputing, so is in midi branch.

There is much more fun to connect such devices, than ordinary mouse or keyboard.
Thanks in advance.

hathach, 2021-05-20:
yeah, I agree. I am improving host stack and its driver, will definitely try to get at least PS4 controller since it seem easy enough. Though I am overwhelmed by current amount of work, so it may take a bit of time.

jhusak, 2021-05-20:
Thanks for quick anwer :)
I use Pi Pico. A lot of fun with it :)
I think PS4 will break through, utilizing others will be much easier then :)

dshadoff, 2021-06-30:
Once this is in motion, I can also help test with various similar-style controllers - also on Pi Pico (hopefully after mouse code is merged into base).

Wikingus, 2021-08-17:
Hello
I join the request.
We need a good library to support a PS4 Pad for Rp pico with Bluetooth support.
A library that supports more than just buttons, but also colors and vibration.
I was about to start writing one, but I have no idea about tinyusb, I just tried to learn something, but at the moment I have no knowledge to write such a library with HOST USB and BT support.

hathach, 2021-08-25:
host hid controller is added with ps4 dualshock support here https://github.com/hathach/tinyusb/pull/1047 . 
video demo https://twitter.com/tinyusb/status/1430406301553594372

For more advance feature e.g set LED. vibrate motor etc check out linux driver for hint https://github.com/torvalds/linux/blob/master/drivers/hid/hid-sony.c

I will close this issue for now, midi host driver will take very long time since there is no plan (and no time) for it at the moment.

closed 2021-08-25 by hathach