hathach/tinyusb #843: MIDI: race condition between device enumeration/configuration and packet read

kaysievers, 2021-05-21:
Issuing a MIDI packet read/receive from the main loop disturbs the USB device enumeration/configuration. The host reports the following:
```
[   74.733249] usb 1-6: new full-speed USB device number 7 using xhci_hcd
[   74.849035] usb 1-6: Device not responding to setup address.
[   75.051975] usb 1-6: Device not responding to setup address.
[   75.260382] usb 1-6: device not accepting address 7, error -71
```

Blocking in the device's main loop until the USB connection is established, or adding a delay of ~200 milliseconds works around the issue.

Checking `ep_out` seems to prevent the failure:
```c
static void _prep_out_transaction (midid_interface_t* p_midi)
{
  if (p_midi->ep_out == 0)
    return;

  uint8_t const rhport = TUD_OPT_RHPORT;
  uint16_t available = tu_fifo_remaining(&p_midi->rx_ff);
```

hathach, 2021-05-22:
can you provide your setup along with way to reproduce the issue ? Please make sure you use the latest code from master branch as well.

kaysievers, 2021-05-22:
It's running on SAMD51, the code is identical to the master branch. It appears pretty obvious that the current implementation does not protect itself from the race explained above.

kaysievers, 2021-05-22:
Here is an older similar report:
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/86#issuecomment-569182562

hathach, 2021-05-22:
@kaysievers can you provide a simplest example along with all the info that could help to reproduce the issue.

kaysievers, 2021-05-22:
Here is a minimal Arduino sketch which produces the failure. Blocking in setup() until USB is ready is not an option, the device needs to work without USB connected too. And it would also fail the same way if the host triggers a reset/reenumeration of the connected device.

```c
#include <Adafruit_TinyUSB.h>

Adafruit_USBD_MIDI usb;

void setup() {
  usb.begin();

  // Add this to make it work
  //delay(500);
}

void loop() {
  uint8_t packet[4];
  usb.readPacket(packet);
}
```

This is the host's kernel log receiving corrupted messages from the device:
```
[  250.428136] usb 1-6: new full-speed USB device number 38 using xhci_hcd
[  250.440293] usb 1-6: device descriptor read/8, error -71
[  250.558831] usb 1-6: device descriptor read/8, error -71
[  250.665745] usb usb1-port6: unable to enumerate USB device
```

theacodes, 2021-06-20:
I've reproduced this on the SAMD21 as well, and I can confirm the patch of checking `ep_out` resolves the issue (at least for us).

hathach, 2021-06-20:
Thanks for confirming, just fixed this, I kind of forgot about this somehow. My memory is fading :sweat:

closed 2021-06-20 by hathach

theacodes, 2021-06-20:
No problem at all, thanks for the quick fix. :)

hathach, 2021-06-20:
also thank @kaysievers for raising and posting patch for this.

PR #906 (2021-06-20): fix issue when calling midi API when not enumerated yet
**Describe the PR**
Fix #843 , add check to prevent issue when calling midi API when device is not enumerated. Which could cause incorrectly submit transfer to endpoint 0x00 (when not initialized).