hathach/tinyusb #851: Migrate example build to CMake ?

hathach, 2021-05-26:
**Is your feature request related to a problem? Please describe.**
CMake seems to make thing more manageable and modulo than plain make. Would it be benefit to migrate example building and adding an CMakelist.txt to src for faster integration.

**Describe the solution you'd like**
`make BOARD=` should still work as it is, the target will thin wrapper for running cmake with build-BOARD and execute make.

cr1901, 2021-05-26:
I'd rather use meson than CMake- a bit of a better cross-compilation story IME. But adding toolchains beyond MSVC, Clang, or gcc is more difficult. So it's not unconditionally better.

It's your call :D!

hathach, 2021-05-27:
Thank you for your suggestion, to be honest, I wouldn't want to move a finger to change something that already works :smile: . Though recently both Espressif and RPi release sdk that support tinyusb, and both use CMake as build tool. I have already learned CMake the ugly way (mostly copy & paste from all over the places). Although It works for now, I get to know CMake better and would like to make the tinyusb integration more smoothly on those platform. 

A quick google suggest meson is a better build tool though, however since I already know CMake and others are using it. I think the decision is more toward the convenience.

cr1901, 2021-05-27:
Well, at least [cmake-microchip](https://github.com/Elemecca/cmake-microchip) exists if a PIC port of TinyUSB ever happens...

hathach, 2021-05-27:
> Well, at least [cmake-microchip](https://github.com/Elemecca/cmake-microchip) exists if a PIC port of TinyUSB ever happens...

Thank you very much for your pointer, that is great to know. Will check if I ever work on PIC port. Every silicon vendor should provide easy-to-use and easy-to-integrate cmake/make that just link bsp. mcu driver for user. Nah, I don't think they care  :sweat_smile: :sweat_smile:

cr1901, 2021-05-27:
Vendors don't get their SVDs/register maps right. What makes you think they'll get the build system right? :D

Daniel-Trevitz, 2022-10-05:
@hathach, I've been working on this as I use cmake for all my projects.
I've got it working, including support for generating tusb_config.h
There are issues with the RNDIS support because of  the dependency on lwip and lib/networking that I could use feedback for.
I also have not integrated into the existing cmake example projects yet; that would be a future change dependent on this.
Are you interested in reviewing a WIP pull request for this?

hathach, 2024-06-06:
most of example are now builtable with cmake

closed 2024-06-06 by hathach